ole00/afterburner #2: Questions about your programmer ~~

sebi5361, 2020-04-26:
Thank you for your great work! It made me want to start playing with GALs.
I have a few questions about your programmer:
- What should be the voltage for an ATF22V10 (10V, same as for ATF16V8)?
- I ordered ATF22V10C (not ATF22V10B). Will afterburner still work with those GALs?
- What is the 7 segment display used for on your design?

Thanks

ole00, 2020-04-26:
Hi, thanks, It's not just my work, it's based on other open-source projects.
 
> What should be the voltage for an ATF22V10 (10V, same as for ATF16V8)?

So far it looks like Atmel works the best with 10V. So, yes, try with 10V programming voltage first.

> I ordered ATF22V10C (not ATF22V10B). Will afterburner still work with those GALs?

It should. It's based on ATFblast software (http://www.bhabbott.net.nz/atfblast.html) which claims it does work, so Afterbuner should as well. Let me know whether it does work for you or not.

> What is the 7 segment display used for on your design?

That's the voltage meter. It's not strictly required if you are happy to measure progamming voltage (
VPP, 10V for ATFs) by multimeter. During development I was trying different GAL chips, therefore voltage meter is handy. The part was bought on ebay (https://www.ebay.co.uk/itm/2-5V-30V-DC-Voltage-Digital-0-36-12V-Two-Wire-White-LED-Meter-Voltmeter-Bike/264685988300) and the mounting holes on the PCB should fit nicely.

sebi5361, 2020-04-26:
Thanks a lot for your answer.
To stay on the safe side, I cancelled my order of ATF22V10C GALs.
Instead I bought ATF16V8B GALs that should fit my needs at that you already tested.
I will start by implementing your circuit on a prototype PCB and use an external voltmeter to set the correct voltage.
I will let you know how it goes as soon as I receive my GALs. Thanks.

closed 2020-04-26 by sebi5361

sebi5361, 2020-04-26:
By the way, [molej.cz/index_aft.html](http://molej.cz/index_aft.html) the website mentioned in afterburner seems to be down...
In the documentation, you mention WinCupl to generate .jed files. Does it work for non-ATMEL GALs?

ole00, 2020-04-27:
> By the way, molej.cz/index_aft.html the website mentioned in afterburner seems to be down...

Ah! thanks for the reminder. I planned to make a web page, then forgot about it. 

> In the documentation, you mention WinCupl to generate .jed files. Does it work for non-ATMEL GALs?
As far as I know it should work. You just have to select corresponding Atmel GAL chip (16V8 or 22V10) , compile the script and the produced jed file can be used on other GAL chip brands.