ole00/afterburner #3: ATF16V8B strange behavior

PerroLoco64, 2020-05-22:
Don't know why but the chip just doesnt work as supposed, only simple equations works (for example when only one gate type is used for each output), 3.6V outputs instead of 5V, only the low signals are really 0V, also I have to config WinCupl in g16v8ma mode if not the chip almost does nothing, even if everything in the process of programing this ICs seems to be ok, reading the fuse map seems to be the same that is in the jed file only the last 6 lines of every single one is kinda diferent, example:

JEDEC file:
*L02048 01000000001100000011000000100000
*L02112 00000000111111111111111111111111
*L02144 11111111111111111111111111111111
*L02176 111111111111111111
*C14B5
*9091

Read from the chip:
L2048 01000000*
N UES 0C 0C 04 00 00 00 00 00*
L2056 0011000000110000001000000000000000000000000000000000000000000000*
L2120 11111111111111111111111111111111111111111111111111111111111111111111111111* 
N PES 31 42 38 56 36 31 46 31*
C14B5
*  

I follow the instructions to program these, 10V, erase first, also tried the conditioning jed file from ATFblast for the ATF16V8B but in this case afterburner returns an error saying that the fuse map wasn't uploaded. Must say that in the moment I ran out of 4.7K resistor, so I used 10k and also tried 1K to see if there was any diference but it was the same, tried the x64 exe in windows 10, the linux executable in debian running under virtual box, even compiling with WinCupl in a windows 98 VM, all were the same results.

Also after some research I found that it was something kinda common for the Atmel PLDs, only GALs from lattice were working (with universal programmers as the TL866).

Did you have the same issue?

Sorry if my english is bad.
Thanks in advance

ole00, 2020-05-23:
Hi,

> reading the fuse map seems to be the same that is in the jed file only the last 6 lines of every single one is kinda diferent,

That should be OK. The last number in the original JED file is an 'end of transmission' character (0x03) followed by transmission checksum, which afterburner does not calculate. But that should not affect functionality. Some software require transmission checksum to be present in the JDEC file, in such case using value 0000 will do the job (it means ignore transmission checksum).

As far as I can tell, if you are able to read back all fuses from the IC and the contents match then the device is correctly programmed. My understanding is that pull down resistors ensure that the pins don't toggle during programming (either randomly or as a consequence of the previous logic programmed in the device) for devices that are sensitive to such changes. It looks like Atmel devices do not require pull downs (see Bruce Abbott's notes linked on the front page), so using 10k or 1k pull-downs should probably be OK in your case (using Atmel IC).

> also I have to config WinCupl in g16v8ma mode if not the chip almost does nothing

In Wincupl sepcify:
`Device    G16V8;`
in your PLD design file.
And in Menu->Option->Devices specify DIP | Atmel | ATF16V8B. That's how it works for me OK.

I'm not quite sure what you mean by 'the chip almost does nothing'. Can you exercise AND | OR | XOR | NOT combinations on several inputs and feed the output to several output pins? Does that work fo you? I think that's pretty much you can do with these 16V8 devices. You can also synchronise the output via register based on input clock pin 1 (only on that one) like this:
```
Device    G16V8;
/**  Inputs  **/
PIN 1  = clock;                /* Register Clock    */
PIN [2,3,4]  = [D0..2];        /* Data Input        */
/**  Outputs  **/
PIN [14, 15, 16] = [Q0..2];    /* Register Output   */
/** Logic Equations **/
[Q0..2].d   =  [D0..2];        /* note .d is used */    
```

> Also after some research I found that it was something kinda common for the Atmel PLDs, only GALs from lattice were working (with universal programmers as the TL866).

I don't know about that (might be true though for certain batches or revisions of the chip). Could you link some sources that suggest that?

Just a silly question: do you test your programmed Atmel chip outside of the afterburner programmer, right? So, do you take it out of afterburner and place it in a bread board, or place it to your testing protoboard? Do you have decoupling capacitors installed on VCC when testing on your test board/pcb? I'm sure you do that, I'm just ensuring there is no misunderstanding what afterburner can do and can not do.

If you have a .PLD script you would like me to try, feel free to post it here and I can test it with ATF and Lattice chip. Just make sure the script is compatible with G16V8 devices.

> Sorry if my english is bad.

Your English is very good !

PerroLoco64, 2020-05-24:
Wow, thanks for the fast answer

> I'm not quite sure what you mean by 'the chip almost does nothing'

When I only write g16v8 in device is when almost all the combinations are wrong, almost all the combinations remains HIGH, when I specify g16v8ma in WinCupl it works a bit better but around the half of the output from the true table stills wrong

> Can you exercise AND | OR | XOR | NOT combinations on several inputs and feed the output to several output pins? Does that work for you?

Yes it can... well actually noticed that not always, my first test was AND OR gates and they worked but now I tested using a not gate and didn't work, the signal wasn't inverted, I mean, a LOW input was a LOW output and a HIGH input was a HIGH output. Until now I haven't tried clock signals or equations with XOR operators

> I don't know about that (might be true though for certain batches or revisions of the chip). Could you link some sources that suggest that?

Here: https://atariage.com/forums/topic/246552-question-about-programming-galplds/
Extrated from there: `These chips were new, however, were purchased from the asian market. Not to mention that this type of PLD is quite old. This leads me to suspect that these are refurbished, or counterfeit. I suppose a small chance there may be of the algorithm not being solid enough to program EVERY IC. But I find that extremely hard to believe.`
I purchased mine where I live, México, through a Latinoamerican page (Mercadolibre with a trusted seller), wouldn't be weird that his products were dispached from china

> Just a silly question: do you test your programmed Atmel chip outside of the afterburner programmer, right? So, do you take it out of afterburner and place it in a bread board, or place it to your [...]

Yes to all of that,

Maybe is a bad compilation from WinCupl, but the 3.6V read is normal? also in the inputs and outputs that are not used the multimeter reads 3.6V too

> If you have a .PLD script you would like me to try, feel free to post it here and I can test it with ATF and Lattice chip. Just make sure the script is compatible with G16V8 devices.

Could you try this? is using only NOT operators, the behavior of this when I burn it into the chips is the same described above, there is also the JEDEC file to compare if your Wincupl compilation is different
[TEST_NOT.zip](https://github.com/ole00/afterburner/files/4672995/TEST_NOT.zip)

Do you think that maybe these PLD I have are just bad? I purchased five ATF16V8B, the behavior is the same in the five chips (using the same JEDEC file)

Thanks again

ole00, 2020-05-24:
I have tested your example on ATF16V8B and Lattice GAL16V8D and you are correct. I can reproduce the issue. When the JED file is uploaded to Lattice GAL16V8D is works as expected - NOT gates are functional. When uploaded  to ATF16V8B the chip does not work as intended.  I can write and read the fuses on ATF16V8B, but so far it looks like the chip interprets the fuses incorrectly. 

> Do you think that maybe these PLD I have are just bad?

I don't think they are bad. I think there is a bug in afterburner. I did not have much time to investigate what is going on, just briefly checked atfblast (the original source code) whether I could spot a bug that I may have introduced during porting the code to afterburner. So far I could not find anything, but will try to resolve it.  

I'll try the same test on ATF22V10C - as they are the same brand (but different programming algorithm and different layout of fuses) and will share my findings.

Apology for any inconvenience. I'll take a look and hopefully find the solution. 

I'll probably need an ATF16V8B with functional logic burned in and corresponding JED file (not read from afterburner). Then I could read the fuses with afterburner and compare the results (fuses in JED files). Hopefully this will give me a clue what is wrong, but it will take weeks to resolve I'm afraid.

PerroLoco64, 2020-05-24:
> Apology for any inconvenience. I'll take a look and hopefully find the solution.

Yeah, don't worry, taking long time is something to expect when things have to be reverse enginered, even more if they are discontinued, there is a lack of info

These PLDs were referenced in a homework documend, the teacher said that wasn't important but I looked into these things because they looked really interesting and are even breadboard friendly, not as CPLDs or FPGAs, 

Was wondering if you tested the Atmel ICs before? I mean, the bug its not caused by a update of something like the version of arduino IDE? I will also to try a look but I'm not too good a programing yet

Thanks for taking the time to answer anyway

PerroLoco64, 2020-05-25:
Hey, just for testing I created a JEDEC file using instead ispLEVER from lattice, the program should do the same that the one from WinCupl: aplying NOT to a input from pin 3, and the output are the pins from 12 to 18, no idea why they are different

JEDEC from WinCupl:
```
CUPL(WM)        5.0a  Serial# 60008009
Device          g16v8s  Library DLIB-h-40-9
Created         Sun May 24 00:28:27 2020
Name            TestNOT 
Partno          00 
Revision        01 
Date            23/05/2020 
Designer        ADMO 
Company         BUAP 
Assembly        None 
Location        
*QP20 
*QF2194 
*G0 
*F0 
*L00256 11111011111111111111111111111111
*L00512 11111011111111111111111111111111
*L00768 11111011111111111111111111111111
*L01024 11111011111111111111111111111111
*L01280 11111011111111111111111111111111
*L01536 11111011111111111111111111111111
*L01792 11111011111111111111111111111111
*L02048 01111111001100000011000000100000
*L02112 00000000100000001111111111111111
*L02144 11111111111111111111111111111111
*L02176 111111111111111110
*C2418
*A056
```

JEDEC from ispLEVER
```
ispLEVER Classic 2.0.00.17.20.15 Lattice Semiconductor Corp.
JEDEC file for: P16V8AS V9.0
Created on: Sun May 24 19:30:55 2020

*
QP20* QF2194* QV0* F0*
 X0*
NOTE DEVICE NAME: GAL16V8Z-12QP*
NOTE Table of pin names and numbers*
NOTE PINS x:6 y:5 z:4 w:3 a:18 b:16 c:17 d:15 e:12 f:14 g:13*
L0256 11110111111111111111111111111111*
L0512 11110111111111111111111111111111*
L0768 11110111111111111111111111111111*
L1024 11110111111111111111111111111111*
L1280 11110111111111111111111111111111*
L1536 11110111111111111111111111111111*
L1792 11110111111111111111111111111111*
L2128 1111111111111111111111111111111111111111111111111111111111111111*
L2192 1*
C236D*
9467

```

On ispLEVER I selected GAL16V8D as device, once compilated burned it into the ATF16V8B trough afterburner and it kinda worked, 4 were correct and 3 weren't, the output pins that worked were 18, 17, 12, 13.

Here is the project from ispLEVER with .jed file:
[test.zip](https://github.com/ole00/afterburner/files/4674788/test.zip)

Hope this helps you someway

mecparts, 2020-05-25:
I think I'm seeing the same (or a very, very similar) issue. I just built up the programmer today and tried it out with an old Atmel AT16V8BQL I had on hand. At first everything looked like it was working. The programmer ID'd the chip, could read from it, erased it, and could write to it. Great! Great that is until I actually tried to use the chip. There doesn't appear to be any output happening. Every I/O pin reads about 3.8V volts. I tried with a known working Jedec file from an old project, and with a simple 7 segment decoder I threw together tonight. Neither does a thing when programmed into the chip.

mecparts, 2020-05-25:
From the "may or may not be useful, but here's a data point" department: I programmed a trivial file consisting of equations like "outX = inA;" and "outY = !inA;". That worked - but in inverse. In other words, "outX = inA;" behaved like "outX = !inA;" and "outY = !inA;" behaved like "outY = inA;".

mecparts, 2020-05-25:
Aha! I think I found it. In afterburner.ino, the galinfo struct used cfgV8AB for the ATF16V8B. In the code, in readOrVerifyGal and writeGal, the code used cfgV8 when it was handling the ATF16V8B. I changed all 3 occurrences to cfgV8AB, uploaded the sketch and programmed a chip. Which worked like a charm. Whew! I've sent a pull request with the changes (it's all of 3 lines) for your perusal.

ole00, 2020-05-25:
> Aha! I think I found it. .. I changed all 3 occurrences to cfgV8AB, ...

Excellent and well spotted! I've applied your fix locally and it indeed fixes the issue. Verified on the ATF16V8B device using PerroLoco64's test PLD script.

>  I've sent a pull request with the changes (it's all of 3 lines) for your perusal.

That's great, many thanks for that.

ole00, 2020-05-25:
I've merged the fix. Thank you mecparts for the fix and PerroLoco64 for reporting the issue.
I' closing this issue, hopefully as resolved.

closed 2020-05-25 by ole00