ole00/afterburner #5: GAL16V8D Reset Afterburner

MD95, 2020-06-01:
After i've programmed my GAL with program which i send below.I cant reprogram, erase, identify ic anymore.Just keep resetting arduino.I've programmed same ic before with simle and, or gates and it worked well without any problem.So i change the arduino, programming voltage but can not solve the problem.Is problem just ic or programmer?

Device g16v8;
PIN 2 = R;
PIN 3 = S;
PIN 19 = Q;
PIN 18 = QN;
Q = !(R # QN);
QN = !(S # Q);

ole00, 2020-06-01:
Hi,
1) which brand and GAL IC do you use? 
2) what VPP (programming voltage) do you use?
Also, your script is interesting: Q is a product of QN and QN is a product of Q. I'm not an expert in PLDs and gate design, to me it looks like a chicken-and-egg conundrum.  But maybe it is not and it is fairly normal expression.

mecparts, 2020-06-01:
After adding in the remaining header elements that CUPL complained about, I still couldn't get it to compile. "pin/node 19 invalid input:  Q" was the error I got.

ole00, 2020-06-01:
@mecparts : thank you for checking, the script looked suspicious to me.

@MD95 : could you also list/attach the full PLD script, the JED file and send the name of the CUPL tool/compiler you are using?

MD95, 2020-06-01:
1. I use Lattice GAL16V8D
2. I tried to implement simple SR latch.Also i'm not sure about my equation because i am beginner and just experimenting.I've tried to program 10v, 11v and 12v not go beyond that.
3.Atmel WinCUPL Version 5.30.4 on Windows 10 64bit

I forget to say i could compile it as g16v8ms.May be is it problem and i fried the IC.

*******PLD File******

Name SRLATCH;
partNo 00;
Date 01/06/2020;
Revision 01;
Designer MD;
Company MD;
Assembly None;
Location None;
Device g16v8ms;

PIN 2 = R;
PIN 3 = S;
PIN 19 = Q;
PIN 18 = QN;
Q = !(R # QN);
QN = !(S # Q);

******JED File*******

CUPL(WM)        5.0a  Serial# 60008009
Device          g16v8ms  Library DLIB-h-40-11
Created         Mon Jun 01 19:17:49 2020
Name            SRLATCH
Partno          00
Revision        01
Date            16/06/2016
Designer        Esteban
Company         estebon
Assembly        None
Location        None
*QP20 
*QF2194 
*G0 
*F0 
*L00000 11111111111111111111111111111111
*L00032 01111111111111111111111111111111
*L00064 11111101111111111111111111111111
*L00256 11111111111111111111111111111111
*L00288 11110111111111111111111111111111
*L00320 11011111111111111111111111111111
*L02048 00000000001100000011000000000000
*L02112 00000000111111111111111111111111
*L02144 11111111111111111111111111111111
*L02176 111111111111111101
*C20A4
*9CB0

mecparts, 2020-06-01:
I thought it was a NOR gate RS flip flop. I'd have never thought of doing it that way (though most of my PLD experience was long ago, far away, and pretty limited).

After changing the device to g16v8ms I was able to get CUPL to compile the file.

The oldest GAL in my parts box is a ATF16V8B from 2003 (I sort of skipped right from PALs to the ATF parts) so the best I could do to check was to write the .JED file out to one of those.

No problem writing, or reading back.

Was able to erase the chip, and read back that it was erased.

Then programmed another .JED file, verified it, and erased the chip again.

I notice that you're using the Atmel WinCUPL. If I recall correctly, that's limited to working on Atmel parts. Does a Lattice GAL16V8 have the same fuse map as an Atmel ATF16V8? If not, that could be your problem (though that's purely a guess on my part). Look in Options -> Devices to see if your version of WinCUPL lists any manufacturers other than Atmel.

ole00, 2020-06-02:
>  Does a Lattice GAL16V8 have the same fuse map as an Atmel ATF16V8?

Yes it does. There might be a confusion that the ATF16V8 can act as 3 different devices, depending on which pins and functions are used. See the datasheet, chapter 10.: 3 modes of operation - Registerd, Complex and Simple. For example in Registered mode pin 11 acts as output enable (OE#), therefore if it is not tied to GND you'll get no output on other output pins. So, this might confuse/trip some people thinking their script/device does not work properly. It's worth reading the ATF16V8B datasheet.

@MD95 thanks for the script, I'll try it soon - I have Lattice GAL16V8 IC available, from the top of my head they require 12V VPP.

MD95, 2020-06-05:
Hello again

I changed my part supplier for my GAL's and there is no problem anymore.I think old ic may be refurbished.So i can easily write and read back again.Thank all of you for support.

closed 2020-06-05 by MD95

ole00, 2020-06-06:
OK, no worries. Thanks for the info. I'm glad it works for you now.