ole00/afterburner #6: Arduino Question

Justinius, 2020-06-27:
I'm interested in building this, but don't have much experience with Arduino.

There is another project that uses a Mega to read/write EEProms. It needs the extra digital I/O. 

Would this just run on the Mega since they seem to have the I/O port in similar places?  Also have you thought about selling this on Tindie or somewhere else?

ole00, 2020-06-28:
> Would this just run on the Mega since they seem to have the I/O port in similar places?

I don't have Mega board, so I can not verify it on the actual board. But looking at the pinout and IO voltage of Mega it should be compatible. Afterburner .ino sketch does not directly access MCU ports, but controls the GPIO via functions that improve compatibility:  like digitalRead() and digitalWrite(), therefore there is a high chance it might "just work".   

> Also have you thought about selling this on Tindie or somewhere else?

These GAL programmers are niche - not many people need those these days, so I think it's not worth doing. Also, I think that building the whole board is more fun compared to just buying the finished product (you can buy a commercial GAL programmer if you want quick&easy solution). Also, for beginners it's a learning experience (producing or ordering PCB, ordering parts, soldering and troubleshooting, putting it all together etc.). For experienced hobbyists it should be a piece of cake that will not take long to build.

Justinius, 2020-06-28:
The BOM only lists one value of resistor. However you have R1, R2, and R3 on the PCB. Are they the same or is the BOM missing some resistors?

ole00, 2020-06-29:
Yes, R1, R2, R3 are the same resistor value 4k7 (I think I also used 5k and it worked OK). The value for R1,R2,R3 is mentioned in schematic.