ole00/afterburner #9: 22V10 Issue

Transistorfips, 2021-05-24:
HI, 
I'm using a different hardware and a different power supply solution to switch Vpp for the GAL (instead of the 5V/12V MT3608 module). Also my Hardware is a Arduino Nano instead o the Arduino UNO. This combo works fine for the Lattice 16V8 (right from the beginning) but the ATF22V10C is not getting recognized.
After hours of drudgening error seeking with the voltmeter and the oscilloscope I wonder if there is a bug in the sw. You sure you have tested your hw with a F22V10CQZ?

Regards,
Joachim

mecparts, 2021-05-24:
Just as a data point, I've been using an afterburner setup to program ATF16V8BQLs, ATF20V10Cs and ATF20V10CQZs for about a year now. And I also use a Nano and a different 10V/5V switching setup.

ole00, 2021-05-24:
I dug out my box with GAL chips and found several Atmel GALs. None of them is the CQZ specifically, but I have ATF22V10C-15PU date code 1809. I must have copied the CQZ from the ATFBlast description, sorry about that. Anyway, plugged the ATF22V10C-15PU in Afterburner and after few attempts to set the voltage right, the IC was identified:
<pre>
./afterburner i  -t ATF22V10C
PES info: Atmel ATF22V10C  VPP=10.0 Timing: prog=10 erase=25
 </pre>

Atmel GALs seem to be quite picky about the right VPP voltage. I dialed  10.35V to make this ATF22V10C-15PU to identify. It seems the VPP depends on the actual value of the VCC. If the VCC is low (for example arduino connected via USB, VCC = 4.534) then the 12V must be lowered accordingly (I suspect there is no voltage reference in ATF ICs, so VPP = VCC * x, in theory x = 12/5). In my case (VCC=4.534) the VPP was set to 10.35 (x=2.282).

I'll try the CQZ version once it arrives (I just ordered one piece on ebay, I should have it in few days, it is sourced locally - not from China.).

ole00, 2021-06-04:
I just got the ATMEL TW F22V10CQZ-20PU 20233WR. It came from China after all, so its origin/authenticity is dubious.
But Afterburner detects it using the same VPP as I wrote in my previous reply. Hope it helps.

thinhx2, 2022-03-11:
> HI, I'm using a different hardware and a different power supply solution to switch Vpp for the GAL (instead of the 5V/12V MT3608 module). Also my Hardware is a Arduino Nano instead o the Arduino [...]

Can you tell me your different power solution?