ole00/afterburner #10: Compile Error (termios.h)

engrjawadulhassan, 2021-06-20:
I am using Win10 x64 with Arduino IDE v1.18.13 , and Ubuntu Budgie With Arduino IDE va.1.18.13

On both sides i am facing this error about "termios.h"

```
In file included from sketch/afterburner.c:52:0:
serial_port.h:114:10: fatal error: termios.h: No such file or directory
 #include <termios.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
termios.h: No such file or directory
```

Any help regarding this will be appreciated.

ole00, 2021-06-20:
Check the compile_win32.sh or compile_win64.sh build script. You will notice both scripts define _USE_WIN_API_ preprocessor symbol, which is used in serial_port.h. Define the  _USE_WIN_API_ in your compilation script or hard-code it somewhere at the top of the serial_port.h and re-run your compilation. That should do the trick.

engrjawadulhassan, 2021-06-21:
Thanks for quick reply , really appreciate this.

I added this line to serial_port.h
`#define _USE_WIN_API_`

and now the error is changed

`In file included from sketch\afterburner.c:52:0:
serial_port.h:7:10: fatal error: windows.h: No such file or directory
 #include <windows.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
windows.h: No such file or directory`

Compiled on Windows 10 x64 under Arduino IDE version 1.8.13

ole00, 2021-06-21:
Well, it looks like there is some confusion here. The afterburner code consists of 2 programs. 

1) One that runs on Arduino itself - that is the afterburner.ino file (and only this file). You can compile that in Arduino IDE and then upload it to your Arduino board - actually this compilation and upload must be done. This code does not use serial_port.h source code therefore should not cause compilation errors you encountered.

2) The other program runs on the PC and communicates with the Arduino board. This software can also be compiled, but not in Arduino IDE. Your error is related to the program #2 which should not be compiled with Arduino IDE. On windows you should be able to compile it with Mingw C compiler, or possibly with the visual studio C compiler (untested). Note that although the program #1 has to be compiled and uploaded to Arduino, the program #2 does not. The version of the program #2 for Windows or MaxOSX can be downloaded already precompiled from the releases/v_0_3 subdirectory of the project. It may be easier for you to use the precompiled binary of the program #2 in this case.

engrjawadulhassan, 2021-06-22:
Hello,

This is the Solution. 

1.Created a New Folder "C Software" and copied 5 files in this folder (afterburner.c , serial_port.h, compile.sh, compile_win32.sh, compile_win64.sh)

2.Then opened "afterburner.ino" file , compiled it and its done

Problem solved.

Actually due to same name of "afterburner.c" Arduino IDE open it along with serial file for compilation , considering these files as dependencies file.

I hope all this conversation will help a newbie like me.

Thanks

closed 2021-06-22 by engrjawadulhassan

ole00, 2021-06-22:
Thank you for the information and the fix suggestion. I moved the afterburner.c into the src_c sub-directory. This should prevent the compilation issue in the future.