ole00/afterburner #12: Programming voltage generator problem

PhilByte, 2021-08-22:
Built a PCB to the schematic but couldn't get the programming voltage to work.
Thought it may be the MT3608 module (modified as instructed) having some issues as other users on the web have claimed. Modified the module to only output 9 to 16V when enabled which worked very occasionally.
Thought some more and realised that the Arduino was often resetting when the 5V supply was switched on and the USB link to PC was interrupted (Beep off/Beep on sounds)
(Arduino is an old type with plug-in mega328)
I think firstly that the USB feed cannot cope with the MT3608 switch-on surge so I added an external 9V PSU. This only helped a little.
I suspect also that the Arduinos's 5V regulator cannot handle the switch-on surge current to the MT3608 module.
I have now added a 7805 regulator from Vin arduino pin to feed the 5V input to this board. (Arduino 5V connection removed) so now the MT3608 supply and the Vcc to the PLDs is independent (except for a diode on the aduino) of the aduino pwer supply using the 9V PSU. The arduino may also use this PSU but can use the USB if necessary. It does mean that the external 9V PSU must be present to use the system. Seems to work OK!
Next version of my PCB will incorporate a 78l05*  to power the PLDs Vcc  and a 7805 with Schottkey diode to give 5.3V to the MT3608 (Should give 5V out when not enabled - compensates for the series diode drop.)
* May need to be a bit beefier, these PLDs can take nearly 100mA when active.
I can now try to program a PLD.

PhilByte, 2021-09-02:
I was trying an ATF16V8B (of dubious origin) but the power problem arose before it was plugged in - just trying to set the voltage initially. 
I have not yet tried using a different Arduino.
I have a TL866ii programmer which normally programs these devices but this one was not co-operating so I wanted to try other set-ups and Vpp settings.
Sorry for the delay - I've been getting my 6502 system up and running. (Hard to find assembler issue)
Phil

ole00, 2021-09-02:
>  the power problem arose before it was plugged in - just trying to set the voltage initially

That's weird. I would not have expect that. I've experienced the buzzing noise when ATF device was plugged in. I made a note in the README troubleshooting section.
<pre>
.. Arduino made a tiny short buzzing noise and then reset itself. What went wrong ?
... most likely the VPP is set too high and the IC does not like that, it pulls the VPP pin down several times
causing the Arduino to reset on brown out. Solution: reduce the VPP voltage by turning the pot clockwise
 on the MT3608 module.
This happens for example on ATF devices when VPP is set to 12V. ATF should use VPP set to 10V when 
programmed by Afterburner.
</pre>
This does not seem to be your case (your ATF is not plugged in), but It looks similar, so maybe it is some sort of short - a flux residue left on the PCB perhaps.

> .. TL866ii programmer which normally programs these devices ..

I did not know the programmer supports ATF gals. Good to know. Thanks for pointing that out.

PhilByte, 2021-09-02:
TL866ii claims to support ATF16V8 to ATF22V10.  It certainly did at least 1 22V10 for me.
Still short on the ATV750 though.
Phil

MarkusPlehn, 2022-07-01:
> Built a PCB to the schematic but couldn't get the programming voltage to work. Thought it may be the MT3608 module (modified as instructed) having some issues as other users on the web have claimed. [...]

how about a small coil between +5Vin MT3608 and Arduino +5V? to limit the initial current?

artgodwin, 2023-01-21:
I also found the problem with turning on the 5V supply to the MT3608. I tested before the board was fully built up and found switching on the 5V switch crashed the arduino program so it stopped responding (I was testing using the arduino monitor screen rather than the afterburner executable). I added 200uF capacitance to the supply BEFORE the switch (so it could more easily withstand the switchon surge) and it mostly worked. Adding the pulldown on the enable pin also seemed to improve things. I don't think its perfect yet but it does mostly work.

Switching off the 5V supply doesn't completely power down the device as some lines are set to a +5 rest state by Vccoff(). So I have mostly inserted the GAL with the 5V supply on, thinking it wouldn't be much worse. I have damaged some GALS but not as a result of this procedure - see my other notes in the bug re GAL20V8A. 

I am using a genuine Arduino Uno powered only by USB, though the USB hub is not the greatest quality and may sometimes provide inadequate current.

ole00, 2023-01-22:
Thanks for the reports. I've isolated the MT3608 and connected is solely (without afterburner) to the Arduino's 5V rail via switch. When I turn on the switch the voltage on the 5V rail dips the same way you reported in your other thread. So the main (or only) contributor to the drip is the MT3608 module. I'll take a look whether it could find an alternative solution.