ole00/afterburner #13: Problem with ATF16V8CZ-15PU and a story

lucadelpollo, 2021-09-27:
Hi guys,

so I just built the Afterburner ( let' see if I can add a pic here ) , this is my version using Arduino Pro Micro and SW I lightly modified to make it work with Platform Io 

[image]

As you can see I went for the route of "an external power supply", no rocket science, plain simple 78S05 with some caps.

I had a bit of a nightmare story with the MT3608 module, I found out for me the simplest way was to lift up pin 4 and solder a small wire , but I also found the set ( 5 ) I bough on Amazon was defective, it was missing a connection between a resistor and the feedback loop making the regulation non-existent ( luckily it was stuck at 4.84V ), you can see a second wire in my pic ( that's the missing connection ).

Anyway so far been trying with 2 chips with the VPP set at precisely 10.0V , an ATF22V10C and an ATF16V8CZ

With the 16V8C I get the following thing ..

afterburner -v -t ATF16V8B i -d COM11

Afterburner v.0.3
opening serial: \\.\COM11
read: 5 ''
opening serial: \\.\COM11
read: 5 ''
opening serial: \\.\COM11
sending 'p' command...
read: 62 'PES info: 3.3V Unknown GAL,  try VPP=10..14 in 1V steps'
PES info: 3.3V Unknown GAL,  try VPP=10..14 in 1V steps
result=0

( what is that read '62' is the ID code ?? )

If I try with the 22V10 instead things seems to go OK ( I yet have to try the chip itself ) however just to let you know this is what I see ..

afterburner -t ATF22V10C i -d COM11
PES info: Atmel ATF22V10C  VPP=10.0 Timing: prog=10 erase=25

JEDEC file for ATF22V10C
*QP24*QF5892*QV0*F0*G0*X0*
L2156 11111111111111111111111111111111111111111111*
L2200 11110111111111111101111111111111111111111111*
L2904 11111111111111111111111111111111111111111111*
L2948 11110111111111111110111011111111111111111111*
L2992 11110111111111111101110111111111111111111111*
L3652 11111111111111111111111111111111111111111111*
L3696 11110111111111111101110111011111111111111111*
L3740 11110111111111111111111011101111111111111111*
L3784 11110111111111111110111111101111111111111111*
L4312 11111111111111111111111111111111111111111111*
L4356 11110111111111111101110111011101111111111111*
L4400 11110111111111111110111111111110111111111111*
L4444 11110111111111111111111111101110111111111111*
L4488 11110111111111111111111011111110111111111111*
L4884 11111111111111111111111111111111111111111111*
L4928 11111111101111111111111111111111111111111111*
L4972 11111111111111111111111111111101111111111111*
L5016 11111111111111111111111111101111111111111111*
L5808 00000000000000001100*
N UES 30 30 20 00 00 00 00 00*
L5828 0011000000110000001000000000000000000000000000000000000000000000*
N PES 33 43 30 31 56 32 32 46 31 FF*
C5EC5
*

And this is the original .JED file I used as output by WinCPUL .. as you can see despite "verify" and all is O.K. the file looks a bit different ( look at those lines of '000000' ) is that normal ?

CUPL(WM)        5.0a  Serial# 60008009
Device          g22v10  Library DLIB-h-40-1
Created         Sun Sep 26 15:57:05 2021
Name            VT100 
Partno          00 
Revision        01 
Date            31/07/2021 
Designer        Engineer 
Company        XXXXXXXXXXXXXXX 
Assembly        None 
Location        Wales 
*QP24 
*QF5892 
*G0 
*F0 
*L02144 00000000000011111111111111111111
*L02176 11111111111111111111111111110111
*L02208 11111111110111111111111111111111
*L02240 11110000000000000000000000000000
*L02880 00000000000000000000000011111111
*L02912 11111111111111111111111111111111
*L02944 11111111011111111111111011101111
*L02976 11111111111111111111011111111111
*L03008 11011101111111111111111111110000
*L03648 00001111111111111111111111111111
*L03680 11111111111111111111011111111111
*L03712 11011101110111111111111111111111
*L03744 01111111111111111110111011111111
*L03776 11111111111101111111111111101111
*L03808 11101111111111111111000000000000
*L04288 00000000000000000000000011111111
*L04320 11111111111111111111111111111111
*L04352 11111111011111111111110111011101
*L04384 11011111111111111111011111111111
*L04416 11101111111111101111111111111111
*L04448 01111111111111111111111011101111
*L04480 11111111111101111111111111111110
*L04512 11111110111111111111000000000000
*L04864 00000000000000000000111111111111
*L04896 11111111111111111111111111111111
*L04928 11111111101111111111111111111111
*L04960 11111111111111111111111111111111
*L04992 11111111110111111111111111111111
*L05024 11111111111111111110111111111111
*L05056 11110000000000000000000000000000
*L05824 11000011000000110000001000000000
*C5EC5
*46E7

Thanks a lot for any help, btw I know very little about how to use GitHub ( I am used to SVN ).

Any help appreciated, cheers.

ole00, 2021-09-28:
Hi, thanks for the report. It looks like a neat build. To answer your questions:

* the number after the "read:" is the total characters read from the serial line. For example "read: 5" means it read 5 bytes from the arduino device. It's really just a debugging info in case the serial connection is not working properly.

* The JDEC format in both files seems to be correct even though it is not "styled" identically. Important part is the checksum C5EC5 at the end of the file, which matches on both of the bitstreams - that means the contents is most likely the same (16bit CRC is not a 100% guarantee the contents is the same though). You can also check bit by bit (see those 0 and 1's in the jdec files) and verify them manually if you want to be sure.

* Related to why ATF22V10C works and ATF16V8CZ does not - I don't know. AFAIK nobody reported / investigated  the C version of ATF16V8. If you look into the afterburner's arduino code the C version of **ATF22V10** reverses the order of bits of the address (see sendAddress() function) compared to B version. It may be something similar for the C version of ATF16V8. Try ATF16V8 B version of the IC to verify your Afterburner device/build is correct (for functionality of 16V8 devices), then try to reverse the bit order of the address sent to the the C version of ATF16V8 (if you have time and would like to experiment).

lond131421, 2024-06-03:
ATF16V8CZ still cannot support it, I found that the program written in does not match the source program.

lond131421, 2024-06-03:
When ATF16V8CZ uses afterburner to read the generated Jed file and then brushes it back, there will be a problem of different programs before and after. If only the ATF16V8CZ program is flashed, it is normal; I used a professional programmer to read the source program Jed file of ATF16V8CZ, and it is normal to write it in afterburner.

lond131421, 2024-06-03:
I am Chinese, please understand if my English is not good.

ole00, 2024-06-03:
@lond131421 : can you give an example of the JED file, that does not work for you?

lond131421, 2024-06-03:
hello：
 I used afterburner to perform read, erase, and write functions on ATF16V8CZ; The file named test.jed is read and retrieved. Please refer to the data read before erasing and the data written to test.jed after erasing, which is completely incorrect. When checking the chip information, the Atmel ATF16V8CZ model did not appear. Finally, I erased the chip and rewrote it to ATFB11. jed (the original file of the chip), and after reading it, it returned to normal.      
您好：
我通过afterburner进行了对ATF16V8CZ的读取、擦除、写入功能；读取出文件命名为test.jed，请看在擦除前读取的数据和擦除后写入test.jed的数据，完全不正确。  还有在查询芯片信息的时候，并没有出现 Atmel ATF16V8CZ这个型号。最后我擦除芯片重新写入ATFB11.jed（该芯片的原始文件），读取后正常。

lond131421, 2024-06-04:
我发现一个问题
读取ATF16V8CZ  程序应该是   ./afterburner r -t ATF16V8B &gt; my_gal.jed  !!!

ole00, 2024-06-04:
> Please refer to the data read before erasing and the data written to test.jed after erasing, which is completely incorrect.&nbsp

1) Please list the contents of the test.jed here, so I can check it with my Afterburner.

>  the Atmel ATF16V8CZ model did not appear.

2) Which parameters did you use when running afterburner to show the model?

> Finally, I erased the chip and rewrote it to ATFB11. jed (the original file of the chip), and after reading it, it returned to normal

3) How did you do erasing writing and reading ? Did you use Afterburner to do that?

4) Do you have the latest version of afterburner installed (both on Arduino and on PC)? If not, which version do you use?

5) Which board/PCB version of Afterburner do you have? 

Please answer all questions 1 to 5, so that I can help you to resolve the issue.

lond131421, 2024-06-04:
我发现一个问题
读取ATF16V8CZ  程序应该是   ./afterburner r -t ATF16V8B &gt; my_gal.jed  !!!
错误写成   ./afterburner i -t ATF16V8B &gt; my_gal.jed  
这是我的错误，抱歉，再次测试，读取是正确的！

I found a problem

Reading the ATF16V8CZ program should be/ afterburner r -t ATF16V8B &gt; my_gal.jed !!!

Wrote incorrectly as/ afterburner i -t ATF16V8B &gt; my_gal.jed

This is my mistake, sorry, test again, the reading is correct!