ole00/afterburner #20: Support for GAL20V8B

mr-who2012, 2022-06-02:
Hi there,
many chinese online-sellers offer the chip "Lattice GAL20V8B" for a few bucks. Is this chip supported by the programmer? The chip is a lot cheaper than the bigger brother GAL22v10.

ismal, 2022-06-06:
I have Lattice GAL20V8B chips from aliexpress and afterburner programs them without any problems (at 12.5V). I also have AE GAL20V8 from ebay and with those I have no luck. At the same time my TL866II programs both, but for those AE I have to set programming voltage 14.5V there. If I do the same with the afterburner (set 14.5V or above) it still does not program. It also does not recognize those chips, returning
PES raw bytes:
00 00 00 00 00 00 00 00 00 00
ER unknown or wrong GAL type (check Power ON)

but even with the '-nc' option it does not work.  Reading those AE chips with afterburner also returns something like this (even though it is properly programmed with another programmer):

JEDEC file for GAL20V8
*QP24*QF2706*QV0*F0*G0*X0*
N UES 00 00 00 00 00 00 00 00*
L2568 0000000000000000000000000000000000000000000000000000000000000000*
L2632 00000000000000000000000000000000000000000000000000000000000000000000000000*
N PES 00 00 00 00 00 00 00 00*
C0000
 
at the same time Lattice 20v8, programmer with the same JED on TL866II produces correction this (with the afterburner and -nc option):

JEDEC file for GAL20V8
*QP24*QF2706*QV0*F0*G0*X0*
L0000 1111111111111111111111111111111111111111*
L0040 1111111111111111111111110111111111011101*
L0320 1111111111111111111111111111111111111111*
L0360 1111111111111111111111111111111111010101*
L0400 1111111111111111111111111111111111101101*
L0640 1111111111111111111111111111111111111111*
L0680 1111111111111111011111111111111111011101*
L0960 1111111111111111111111111111111111111111*
L1000 1110111111111111111111111111111111111101*
L1280 1111111111111111111111111111111111111111*
L1320 1011111111111111111111111111111101011101*
L1600 1111111111111111111111111111111111111111*
L1640 1011111111111111111111111111011111011101*
L1920 1111111111111111111111111111111111111111*
L1960 1111101010111111111101111111111111111110*
L2000 1111111111110111111111111111111111111111*
L2240 1111111111111111111111111111111111111111*
L2280 1110111111111111111111111111111111101110*
L2560 01100001*
N UES 00 00 00 00 00 00 00 00*
L2568 0000000000000000000000000000000000000000000000000000000000000000*
L2632 00001100111011111110000001110000011000001000000111000001100000011111011000*
N PES 00 00 00 00 00 00 00 00*
C5985

This is Lattice, afterburner and no -nc option:

JEDEC file for GAL20V8
*QP24*QF2706*QV0*F0*G0*X0*
L0000 1111111111111111111111111111111111111111*
L0040 1111111111111111111111110111111111011101*
L0320 1111111111111111111111111111111111111111*
L0360 1111111111111111111111111111111111010101*
L0400 1111111111111111111111111111111111101101*
L0640 1111111111111111111111111111111111111111*
L0680 1111111111111111011111111111111111011101*
L0960 1111111111111111111111111111111111111111*
L1000 1110111111111111111111111111111111111101*
L1280 1111111111111111111111111111111111111111*
L1320 1011111111111111111111111111111101011101*
L1600 1111111111111111111111111111111111111111*
L1640 1011111111111111111111111111011111011101*
L1920 1111111111111111111111111111111111111111*
L1960 1111101010111111111101111111111111111110*
L2000 1111111111110111111111111111111111111111*
L2240 1111111111111111111111111111111111111111*
L2280 1110111111111111111111111111111111101110*
L2560 11100110*
N UES 00 00 00 00 00 00 00 00*
L2568 0000000000000000000000000000000000000000000000000000000000000000*
L2632 11111111110000001110000011000000110000001100000011000000111000001100000011*
N PES 00 03 3A A1 00 00 00 C0*
C5752

For the same chip the last line is apparently also different (at L2632)

ole00, 2022-06-07:
Thanks for the interesting report. I checked Aliexpress and I can't find any Lattice branded GAL20V8B ICs for sale. They have some unbranded GAL20V8B ICs. When searching ebay.com I could find several Lattice branded GAL16V8B ICs for sale, but I had no luck finding AE (is it a brand/manufacturer?). So basically my search was opposite what you wrote: I found Lattice on ebay and unbranded gal chips on aliexpress. Is that correct? What is the AE designator?

ole00, 2022-06-07:
I've checked the original source code and schematic from GALBlast and it does support GAL20V8B. Also, ATFBlast mentions this: "GAL20V8 requires a socket adapter to remap the pins.". So in theory support for GAL20V8B could be added to Afterburner.

ole00, 2022-06-07:
@ismal Could you please share the original JED file that you programmed the GAL20V8B on TL866II? I would like to check  whether Afterburner could produce correct/idetical JED file when reading the GAL burned by TL866II. Thank you.

ismal, 2022-06-10:
Indeed the pins have to be remapped, without that nothing will work, but that was done. The original file is like this. It works with afterburner without any problems as well. There is jsut something wrong with reading that AE GAL. I will send you the link in the next post.

 
GAL20V8
EQN2JED - Boolean Equations to JEDEC file assembler (Version V024)
Copyright (c) National Semiconductor Corporation 1990,1991
Assembled from "C:\DOCUME~1\PC\Plocha\OPAL\ZX48TO~1.EQN". Date: 1-4-109

*
NOTE PINS MREQ:1 ZX512:2 A1:3 WR:4 DIS128:5 BANK2:6 A5:7 BANK0:8*
NOTE PINS BANK4:9 BANK3:10 BANK1:11 GND:12 A15:13 A14:14 ROMCS:15*
NOTE PINS CLK7FFD:16 SA18:17 SA17:18 RAMCS:19 SA16:20 SA15:21*
NOTE PINS SA14:22 IORQ:23 VCC:24*
QF2706*QP24*F0*
L0000
1111111111111111111111111111111111111111
1111111111111111111111110111111111011101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L0320
1111111111111111111111111111111111111111
1111111111111111111111111111111111010101
1111111111111111111111111111111111101101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L0640
1111111111111111111111111111111111111111
1111111111111111011111111111111111011101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L0960
1111111111111111111111111111111111111111
1110111111111111111111111111111111111101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L1280
1111111111111111111111111111111111111111
1011111111111111111111111111111101011101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L1600
1111111111111111111111111111111111111111
1011111111111111111111111111011111011101
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L1920
1111111111111111111111111111111111111111
1111101010111111111101111111111111111110
1111111111110111111111111111111111111111
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L2240
1111111111111111111111111111111111111111
1110111111111111111111111111111111101110
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000*
L2560
11100110*
L2568
0000000000000000000000000000000000000000000000000000000000000000*
L2632
11111111*
L2640
1100000011100000110000001100000011000000110000001110000011000000*
L2704
11*
C5752*
0000

ismal, 2022-06-10:
This is what I bought on aliexpress, and now I see that on the picture it is not really lattice but I got 10 chips with a nice label Lattice (I am not sure if those are fake or not) but they definitely work with this afterburner and recognized as lattice with vpp=12.5V

https://nl.aliexpress.com/item/4000975808375.html?spm=a2g0o.productlist.0.0.345c24df5zYiAo&algo_pvid=f2acfe34-4ccb-4f2d-bd58-cf3de1e779df&algo_exp_id=f2acfe34-4ccb-4f2d-bd58-cf3de1e779df-0&pdp_ext_f=%7B%22sku_id%22%3A%2210000013025128511%22%7D&pdp_npi=2%40dis%21EUR%21%214.54%21%21%211.97%21%21%402100bddf16548481453951943e8a90%2110000013025128511%21sea

This is AE from the ebay. I order it because it is cheap delivered to Europe.

ole00, 2022-06-10:
thanks for the source JED, it proves the afterburner could read the data correctly (same CRC found in JED as the original), but can not write at the right CFG locations. I think I could fix that - it seems to be a bug in afterburner not properly reading /interpreting PES data before it writes the configuration fuses (because there is the No-Pes Check applied).

Remapping of the pins: if you have some info / remapping table please post it here. I just received a Lattice GAL20V8 so I could try to make it work. Thanks.

The links were useful - it looks like those AE marked chips were originally produced by National Semiconductor. Interestingly the chip name does not end with B like the Lattice 20v8B
so perhaps that means the protocol or timing (as you suggested) is different.

ole00, 2022-06-11:
I managed to program GAL20V8 with an adapter [see link](https://github.com/ole00/afterburner/raw/master/img/gal20v8_adapter.png) I made (based on GALBlast schematic). It looks like Afterburner supports GAL20V8 just fine without any source code modifications. All I had to do is to specify "-t GAL20V8". 

I saw one issue, that @ismal mentioned - L2560 was not properly programmed. But that can be fixed (at least in my case) by raising the VPP to 12.5V.  Previously my VPP was 11.5 and during erase not all fuse bits were erased (erase and then read to check the bits). When I re-erased then all fuse bits got eventually erased. Once the VPP was raised, the all fuse bits were erased on the first run.

Related to L2632 - this seems to be correct when '-t GAL20V8' is used. Compared to @ismal's original JED,  Config bits starting at address 2632 are merged by Afterburner to one line. In original JED file, the L2632 aslo start with 8 high bits (11111111) then the line is split to L2640 and continues with the res of the configuration bits. Apart from that split (which is only a formatting difference) the data seem correct.

I've updated the schematics with the GAL20v8 programming pinout.

ismal, 2022-06-12:
Thanks for the update! Indeed, I just wired afterburner only for GAL20V8, because I only needed that. I followed the pinout from here, where the pin1 was grounded without any pull down resistors, and the rest of pins with "VIL" there, were connected to the ground with 4k7 resistors: https://k1.spdns.de/Develop/Projects/GalAsm/info/galer/proggal.html

Erasing of that AE chip was not a problem, but now it broke down, I mean it probably reached the number of rewrites. Still, with the TL866 I can read the stuff back, what is recorded there, and with the afterburner it reads only those two lines, something like: 
JEDEC file for GAL20V8
QP24QF2706QV0F0G0X0*
N UES 00 00 00 00 00 00 00 00*
L2568 0000000000000000000000000000000000000000000000000000000000000000*
L2632 00000000000000000000000000000000000000000000000000000000000000000000000000*
N PES 00 00 00 00 00 00 00 00*
C0000

Indeed, the PES information is totally wrong (or probably it is like that in the chip), it contains all the 00, so I had to use "-nc" option

The AE chip is indeed GAL20V8 without "B" but TL866 also has chips without "B" in the database. 

I can probably test reading from that AE chip is there are some modifications to the code, to see if the output is reasonable, but that indeed would involve some changes in the timing. The voltage vpp was +14.50-15.0 and that worked with TL866,. so it should be good. That chip did not work with lower voltages. At the same time, the Chinese Lattice was working at vpp=12.5 on both afterburner and TL866.

ismal, 2022-06-12:
One thing that I remembered actually, when I was erasing that AE chip, I could read all "1" properly in a nice table, still with PES all zero. Only after programming (even with the TL866, which means it was correctly done), reading anything back was producing those 2 lines.

ole00, 2022-06-13:
The link you sent has some good information about GALs, I'll put it on the front page.

Related to timings - I think you might be onto something. The GalAsm document mentions timeout between STR puls (strobe) 5ms. Afterburner currently does ~2ms. If you check afterburner.ino, line 644 function strobeRow() you can see the function calls strobe(2). You can experiment with different values (probably with value 5 as stated in GalAsm) - maybe the 2ms is currently not enough for National Semiconductor GAL chips to return PES data.

ismal, 2022-06-13:
I tried that solution before (with 5) but it did not work. Also on that website that I mentioned before about 20V8 the strobe during the reading is 5 microseconds and not milliseconds. I tried and it also did not work. TL866 does have options for 20V8 and 20V8B and on that website I also see that there is difference like that, but I am not sure if it is relevant:

The Architecture-Control-Word has the following structure (82 Bit wide):

GAL20V8:
Bits  0-31: 32 Bit product term enable 0-31
Bits 32-35: 4 Bit XOR(n) for OLMC pins 22-19
Bit     36: AC0-Bit
Bits 37-44: 8 Bit AC1(n) für OLMC pins 22-15
Bit     45: SYN-Bit
Bits 46-49: 4 Bit XOR(n) für OLMC pins 18-15
Bits 50-81: 32 Bit product term enable 32-63

GAL20V8A,B:
Bits  0-3: 4 Bit XOR(n) for OLMC pins 22-19
Bit     4: AC0
Bit   5-8: 4 Bit AC1(n) for OLMC pins 22-19
Bit  9-72: 64 Bit product term enable PT0 - PT63
Bit 73-76: 4 Bit AC1(n) for OLMC pins 18-15
Bit    77: SYN
Bit 78-81: 4 Bit XOR(n) for OLMC pins 18-15

Also I was wondering, on that website it says "By clocking SCLK with a
LOW-HIGH-clock transition all the various bits are sent out the SDOUT pin." In the current version of .ino we have 
static char receiveBit(void)
{
    char b = getSDOUT();
    setSCLK(1);
    setSCLK(0);
    return b;
}

apparently it works, but shouldn't it be something like, so we clock and then read the bit:
static void sendBit(char bitValue)
static char receiveBit(void)
{
    setSCLK(0); 
    setSCLK(1);
    char b = getSDOUT();
    return b;
}

It is pity that there are no good time diagrams for the programming process with all the possible ranges for the timings and delays.

ismal, 2022-06-13:
Problem solved :))) I managed to find the datasheet for GAL20V8 from national, and one one page "2-36" I found very small note below one of the diagrams on preloading. It says " The SDOUT output buffer is an open drain output during preload. This pin should be terminated to Vcc with a 10K resistor". I did that and everything reads perfectly fine. Even PES is now saying that it is National GAL20V8 (see below). 

Now, I am not sure it the same resistor can be used with Lattice chips (those work even without the resistor). Having only broken AE I cannot test writing part, but I would assume that it works as well.

JEDEC file for GAL20V8
*QP24*QF2706*QV0*F0*G0*X0*
L0000 1111111111111111111111111111111111111111*
L0040 1111111111111111111111110111111111011101*
L0320 1111111111111111111111111111111111111111*
L0360 1111111111111111111111111111111111010101*
L0400 1111111111111111111111111111111111101101*
L0640 1111111111111111111111111111111111111111*
L0680 1111111111111111011111111111111111011101*
L0960 1111111111111111111111111111111111111111*
L1000 1110111111111111111111111111111111111101*
L1280 1111111111111111111111111111111111111111*
L1320 1011111111111111111111111111111101011101*
L1600 1111111111111111111111111111111111111111*
L1640 1011111111111111111111111111011111011101*
L1920 1111111111111111111111111111111111111111*
L1960 1111101010111111111101111111111111111110*
L2000 1111111111110111111111111111111111111111*
L2240 1111111111111111111111111111111111111111*
L2280 1110111111111111111111111111111111101110*
L2560 11100110*
N UES 00 00 00 00 00 00 00 00*
L2568 0000000000000000000000000000000000000000000000000000000000000000*
L2632 11111111110000001110000011000000110000001100000011000000111000001100000011*
N PES 00 01 20 8F 00 00 00 00*
C5752

PES raw bytes:
00 01 20 8F 00 00 00 00 00 00
PES info: 5V National GAL20V8  VPP=13.75 Timing: prog=80 erase=25

ole00, 2022-06-14:
Well done making it to work. Pull up on the SDOUT was pretty tricky! None of the GAL chips I tested so far required it. I'll add the 10k pull up and will see how it works with Lattice and Atmels. V8 and V8B's "Architecture-Control-Word" is so called Configuration bits in Afterburner - offsets for these bits are stored in arrays 'cfgV8' and 'cfgV8AB'. And there is a bit of code that selects the right array depending on the data from PES. The code is like this:
<pre>
if (pes[2] == 0x1A || pes[2] == 0x3A) {
  cfgArray = (unsigned char*) cfgV8AB;
}
</pre>
In your case pes[2] is 0x20 therefore the bob-B version of the config should be used - which is correct I'd say.

ole00, 2022-06-14:
I checked ATFBlast and GALBlast schematics for the SDOUT signal and they both do have pull-up resistor (4k7) on SDOUT. It is the ACK pin 10 on the Printer (LPT) port.  I think you found a genuine HW bug in Afterburner - well done! I have not put the resistor on my board in yet, but given the ATFBlast and GALBlast both have it there it doubt it will make any issue. I will try it though and update the designs. Thanks again.

ismal, 2022-06-14:
Good to hear that it was helpful! 

I found that note in this datasheet (page 2-36) https://www.amigawiki.org/lib/exe/fetch.php?media=de:parts:gal_nsc_20v8_a.pdf

I checked also a datasheet from Lattice but I have not seen such notes. Also from the behavior (that it actually works with the afterburner) most likely those do have a pullup resistors build in. On the first page in those datasheets it does say "ACTIVE PULL-UPS ON ALL PINS" but somehow for national it was saying that especially for "preload" that SDOUT starts floating.

Taking into account that TL866 worked with both brands in my case (AE (National) and Lattice) and that AE is not even in its database (I was choosing Lattice GAL20V8 to program that National), TL866 should do external pulling up as well. Even "doubling" pulling up for the actual Lattice chips should do no harm with those 10k resistors.

ismal, 2022-06-15:
Just tested and indeed it works with Arduino's internal pullup on SDOUT!

mr-who2012, 2022-06-22:
Wow, 17 comments! Thanks for reply! Since I'm a newbie to GAL's, there isn't much I can contribute to this thread. Just glad I found an interesting topic for everyone.

ole00, 2022-06-22:
I would still recommend to stick to 16V8 or 22V10 GALs. It seems that my Lattice GAL20V8B stopped working for some reason (Afterburner can no longer identify), so this topic will need more investigation. I ordered some additional Lattice 20V8 GALs (from China, they may be clones though) and will try them again later.

ismal, 2022-06-26:
In my experience it works good so far with GAL20V8. There is no difference in terms of programming protocol between 20v8 and 16v8 I believe. I also killed one of my National GAL20V8 but I think it is just because I was erasing and reprogramming it too many times. I read that those GALs can be reprogrammed max 200 times and retain the data for max 20years. I did not do 200 times, but definitely 50-70, so probably it just died:)

mr-who2012, 2022-11-18:
Can you please write more about the GAL20V8B-Adapter? There is just this png-file with missing values of the resistors. I am trying to redraw a pcb in kicad - and it's my first time.  Thanks in advance.

ole00, 2022-11-19:
The pcb design for the adapter already exists in pcb/adapter20v8.pcb. You can open in in the GedaPCB free cad design tool. Please be ware the adapter - as is in the pcb design file - was not tested yet.
The schematic schem_gal22v10.png that is on the main project page has the pinout for GAL20V8, and states that "Pins without a label are pulled down with a 4k7 resistor".

ole00, 2022-11-19:
Here is my adapter prototype made on a proto-board. The real adapter board is the same, it is just more compact. The image shows how the pins and sockets are soldered. Also note that I used  socket and pins with circular pin/hole profile rather than the normal square ones.
[image]

Hope it helps.

mr-who2012, 2022-12-01:
Thank you very much! I was using an old clone of the repo where this file wasn't included. In the current version it is present. I just received the pcb from china and can't wait to assemble the programmer. For the adapter, I'll try your proto-board-attempt. Thank you for the photos!