ole00/afterburner #21: simple GUI desired ?

MarkusPlehn, 2022-07-09:
a pal of me has made a simple GUI for your project. Do you want to have this tool in your repository ?

ole00, 2022-07-09:
Thanks for the info. I'm not a GUI fan, so probably not. But I'm happy to put a link on the main page (Readme.md) so his/her project can be easy to reach.

Dale-Lien, 2022-12-17:
@MarkusPlehn Hi, I want to use the  GUI, can you share it?

MarkusPlehn, 2022-12-17:
Sorry, but the GUI does not work correctly and the creator have no time to
fix it.
So, no GUI for public

ole00, 2023-02-01:
@quantumDavey Creating a GUI for a console application is not that hard. Maybe take it as a challenge and try to make it yourself. Use programming language that is simple (does not need compilation) and cross-platform (runs on Linux, Windows and MacOS) : for example Python might be a good candidate. Then check for a GUI project written in that language for some other tool (there must be some on Github as well). Pick one that is simple - just one screen with a butto, file selector and a text / console window - and research how it is done. If you get stuck with some code, look for tutorials on the web.