ole00/afterburner #22: Support for GAL20V8A

Fisiekk, 2022-08-29:
So i have few GAL20V8A and wanted to ask if they are supported or is it possible to add support for them

artgodwin, 2023-01-21:
I also have some Lattice 20V8A-15LP I'd like to use. Datecode is L241D12
I have built the programmer with the pin mapping as shown for the adapter and can read the ID OK. It reports 

PES info: 5V Lattice GAL20V8  VPP=14.50 Timing: prog=10 erase=25

However, if I set Vpp above about 13V the GAL is apparently destroyed and the PES can no longer be read.

If I attempt to program at 13V the programming works partially : on my first attempt it reported 2000 or so fuse errors. Erasing apparently works ok. However, repeating the programming operation without erasing programmed a lot more fuses. Repeating several times (about 10) got down to a few tens of errors. The last couple of errors took many attempts, perhaps 20 or 30 but eventually worked. Shortly thereafter a repeated verify failed one fuse again but with a few more repetitions it programmed completely and has so far rretained the programming (next day). I haven't tried to verify the functionality yet.

Is this characteristic of any known problems ? Perhaps faulty GALs, or unstable Vpp ? I will try to examine the programming cycle to see if I can find anything going wrong, and perhaps experiment with longer programming pulses. I've had these devices in stock for a long time and have enough to experiment with; it's unlikely they are counterfeit as they date from before that was common but they could be a poor batch, discarded because of low yield.

artgodwin, 2023-01-21:
[Screenshot_2023-01-21_22-39-00](https://user-images.githubusercontent.com/677657/213889877-921f5595-f494-4fe0-b696-37189cdcf331.png)

Fisiekk, 2023-01-21:
Its always decoupling xD

artgodwin, 2023-01-21:
[image]

ole00, 2023-01-22:
Thanks for the report.

It's interesting what you wrote about re-writing the fuses, but it makes perfect sense that re-writing (with the same bit design) without erase will fix these errors eventually. Flipping the fuse bit (internally inside GAL) requires certain amount of current/voltage and if there is instability then the next bit to flip (especially if there are 2 bits to flip consequently) might not get enough current/voltage to flip the bit properly. Next attempt to write (again speaking of internal logic of the GAL) possibly recognises the bit 1 is already flipped so no current/voltage is applied (and it stays in favourable state to program the next bit, which previously failed flipping), therefore the bit 2 has good conditions to be flipped successfully.

PES info and reported 14.5V:  The VPP value, timing value and erase values are decoded from the PES (chip electronic signature) bytes retrieved from the GAL chip  in the Arduino sketch (with the exception of Atmel GALs). The decoding might be wrong.  It might be worth trying the programming with VPP set to 12V as that's the VPP I normally use with Lattice GALS without issues. If there are still the same writing issues with VPP 12V  as with VPP 13V, the afterburner could be modified to do auto-rewrite (multiple times) if it detects writing errors.

Regarding the scope trace 1 (power dip) - I did not know about that, good that you pointed it out. I scoped 5V rail on my board (scoped on Arduino 5V pin header, powered only via USB) in the same situation and indeed my 5V rail dropped the same way. I tried it with and without GAL chip, both produced the same dip.  Because your main decoupling capacitor is before the switch (and fully charged when the switch-on happens, therefore not contributing to the 5V rail dip) I guess the culprit might be the MT3608 module. Arduino does not seem to be reset when that 5V dip happens. I don't think this should contribute to the writing failures you mentioned though (you might try scoping the 5V rail during writing the GAL fuses, whether there are similar dips on 5V rail). Still worth investigating whether this could be fixed.

Useful design: the main credit goes to Bruce Abbott and Manfred Winterhoff for doing the GAL research and publishing their result and their code.

ole00, 2023-01-22:
Just tested and scoped the MT368 module on its own ad it is the reason why the 5V rail drops when the switch is turned on. No wonder, it has a big inductor.

artgodwin, 2023-01-22:
> Useful design: the main credit goes to Bruce Abbott and Manfred Winterhoff for doing the GAL research and publishing their result and their code.

Yes, I don't underestimate the amount of work there. But those designs have quite a few components and the parallel port dependency makes them awkward. Building onto an arduino and keeping it as simple as possible makes it far more attractive as a quick hack - if it gets much more complex you might as well buy a TL866 and use some of the open source replacement firmware.

For that reason I'd discourage the programmable Vpp. Yes, it would be great .. But unless it's implemented with PWM or some other such cheap solution its going to add a bunch of less common parts and lose the biggest advantage. Even the multiple pinning I suggest elsewhere would be a bad idea if it added more than a few resistors and transistors.

I was also surprised that the 5V switch created such a large drop whilst enabling Vpp didn't. Note that the boost converter isn't enabled at that point - so I guess it has to be the capacitors. That's why I put the 200uF upwind of the switch - I didn't want them having to be charged too.

Interesting that you have found 12V enough. I note that the voltage requested is already 2V less than apparently wanted by the chip's algorithm. I'd speculate that the older chips need higher voltage and are also perhaps more sensitive to overvoltage.

ole00, 2023-01-22:
Complexity of Afterburner and additional features: I completely agree. 

The idea of programmable VPP though would still use MT3608 module, but this time it would be used to supply like 18V which would feed LM317T  variable voltage regulator. The output voltage of that regulator (VPP for GAL) would be controlled by digital POT , something like MCP40D18T. It sound simple, but still requires to source less common parts (at least the programmable POT may not be in everyone's drawer). Anyway, I'm not going to add that feature in the near future.

200uF CAP before the switch - thanks for that. I'll test that and modify the design if it reduces the dip at least above 4V. 

12V as VPP:  this article (https://k1.spdns.de/Develop/Projects/GalAsm/info/galer/proggal.html) distinguishes 2 different voltage values of  VPP for reading and writing. According to that article GAL reading (like PES reading) requires always 12V, but writing requires higher VPP ranging from 14V to 15.75V.  Afterburner can do only one VPP during the single read/program session, so it is actually quite fortunate that many of the supported GAL types can be programmed by Afterburner using only single VPP for both reading and writing. Of course you encountered writing issue on your  20V8A, so this might be the odd case.

artgodwin, 2023-01-23:
The 2nd trace above is WITH 200uf - it's far from enough to stop the 5V rail froim being sucked down. Admittedly this is the Uno with an inline FET in 5V - you did sday the cheap clones had a betyter 5V rail. But I've now added 1000uF along with some fast ceramics and it still drops by 1V. I'll add a capture tomorrow. I should probably move this to the discussion section, it's well off topic for 20V8s now. I couldn't believe the 10uF ceramic caps on the 3608 board were pulling such a spike so I looked a bit further .. it turns out that even though the device should be disabled by the Vpp enable control, it actually spikes to 10V or so when the power comes on, settling back to the expected 5V in 100ms or so. So the 5V switch, presumably there to protect the GAL from connection  mishaps, is actually quite unhelpful.

artgodwin, 2023-01-23:
[image]
[image]

Blue is unswitched 5V, green is switched 5V (and trigger), yelow is 3608 output. The traces are the same thing, just zoomed to show both the length of the high voltage pulse and the detail of the 5V disturbance. I did also check that the enable pin stayed low during the switch-on

ole00, 2023-01-23:
Sadly, mt3608 datasheet states: " The MT3608 has internal soft start to limit the amount of input current at startup and to also limit the amount of overshoot on the output.". I guess that does not work out very well for both of the parameters (current limit and overshoot).

artgodwin, 2023-01-23:
It does seem less than perfect :)

ole00, 2023-03-28:
@artgodwin I re-read your report about your Lattice  20V8A. I checked GALep's definition for that specific GAL. The VPP they use is 12V. So either the PES data as read from the GAL are incorrect (you reported Afterburner prints 14.5V, so in PES it was 16.5V) or they have to be interpreted differently for each vendor perhaps.  It may also be, that the voltage really needs to be above 13V as you tried, but it needs to be ramped up slowly not to destroy the chip (GALmate uses that techinque for ST chips). Also, on your scope trace you can see that the VPP  during ramp-up has high spikes reaching possibly 14V which may be the cause of the damage. Unfortunately that's a downside of using a switching boost convertor. On the new Afterburner design (programmble VPP) I resolved it by progressively ramping up, slowing the steps as the VPP reaches the desired voltage. Also, the new design adds 3 decoupling CAP on the VPP: 10uF, 100nf and 2nF which help to reduce the VPP spikes during VPP ramp-up. The 5V rail dip, although not nice, should not damage the GAL, most likely it causes Arduino to reset.