ole00/afterburner #23: Forum wanted

mr-who2012, 2022-12-09:
Is there a forum connected to this project, to discuss troubles when building the device? My afterburner doesn't identify the GALs anymore. At the beginning afterburner did (but reported the GALs wrong).

Now I don't know where to locate the problem: arduino, afterburner, GAL-device or PC. I recognized the caps on the board are all parallel so they sum-up in value. Since the big cap has a tolerance of 20% (see bom.txt), the two small caps may not have an effect, since they are both very small. So I just tried a very large capacitor but without success. Also, measuring the total capacitance in-circuit, gave me a ten-times bigger value than the capacitors have alltogether. I think it's because the power-line is always surrounded by large ground-areas on the board.

All this is very interesting, but maybe completely unrelated to my problem. I tried additional power to the arduino, I tried all voltages from 10V to 14V for the GAL, and I tried three GAL-chips in total. I only have five of the same type and just wait for new chips to arrive. Next step is a logic-analyzer, to see if the chip is sending different signals than arduino/afterburner. 

Erasing a chip is possible, but not reading and writing.

ole00, 2022-12-10:
Hi,
I've enabled "Discussions" tab - feel free to share your thoughts.

The issue you described - it might be caused by the change introduced in version 0.4 : baud rate changed to 57600. If that is the case, Please re-upload the .ino sketch to Arduino and recompile the pc software. Hopefully that's the fix.

Related to the caps values. I doubt that is an issue, they are decoupling caps with relatively low values, so should not cause a trouble unless they are faulty or damaged.

Diagnosing the issue: you correctly listed possible faults ( arduino, afterburner, GAL-device or PC) , so  start checking one by one and eliminate them. Start from the PC connection to Arduino via regular serial terminal (teraterm, minicom, putty etc.) and try to issue some basic Afterburner's commands from there. Arduino test: try a basic blinky sketch, and exercise all digital GPIOs. GAL device - try Atmel's ATF16V8B, these are still produced (eliminates a possibility of a GAL fault from an old and used batch) so IMO best to start with.

Also, I found the official Arduino UNO (which I got and tested way later) is not the best option for Afterburner. It has very large voltage drop on the 5V rail when powered via USB cable. Also, I had to replace the voltage regulator on it because the voltage regulator failed early on. So far my best results are using cheap asian Arduino UNO clones.

mr-who2012, 2022-12-13:
Thank you for answer. I am using one of the cheap arduino-clones. Once, I attached 12V-Power-Supply to give the arduino more juice. It _should_ convert it to 5V. However I still had a bad feeling and soon switched to a 5V-Power-Supply and hope my arduino-clone is not damaged now. 

I attached a Logic-Analyzer while issuing this command in afterburner:

`afterburner r -t GAL16V8 -d /dev/ttyUSB0 -nc `

I get interesting readings: Arduino is sending address-numbers in 4-bit-steps and correctly switches the GAL to edit-mode (pin2 for edit-mode is not visible in the screenshot). The signals on the /STR-pin also seem to be correct. The clock pulses are always following a falling /STR-pulse to transmit the address to internal register. So that's correct as well. So it seems my arduino is executing the code fine. Not sure if it is important: The whole transmission of adress-numbers is done twice.

[image]

The following screenshot shows the [different assignment](https://k1.spdns.de/Develop/Projects/GalAsm/info/galer/proggal.html). of pins in edit-mode (fore reference). Pin15 (for data-output) is constantly showing HIGH in my test. So that's wrong!

[image]

Also, All the VIL-Pins should be tied to ground in edit-mode. But except for pin1 they are all HIGH. It seems on the board they are not connected to the arduino at all. So maybe it's either a faulty chip or wrong pull-down-resistors? Pin1 is correctly LOW-state and has no pull-down at all on the board.

[image]

I also attached a PDF with the text from the GALer-website. In this version I colored the text and made it more readable. Hopefully it will be of use for somebody else.

[Programming GALs.pdf](https://github.com/ole00/afterburner/files/10216995/Programming.GALs.pdf)

Thank you for the possibility to open discussions. I will switch to it immediately.

ole00, 2022-12-13:
I think you made good progress.

> The whole transmission of adress-numbers is always done twice.

It is not done twice. GAL16V8 has 32 so called 'rows' (or addresses)  which are passed via RA0 - RA5 : that is 6bits - values 0 - 63. Because we have only 32 addresses on GAL16V8 the RA5 should always be 0 (Low) when reading GAL16V8. On your picture you can see ~29 addresses being clocked - the total is probably 32 and that is OK. What seems wrong is
your RAG2 and RAG3 signals - they seem to be duplicates of RAG0 and RAG1. The read command reads fuse bits from each address, and the address is incremented sequentially (see readGalFuseMap() function in afterburner.ino). So RAG2 should look like RAG4 and RAG3 should look like RAG5. Either your probes were incorrectly placed or were touching during measurement (shorted pins).

> Pin12 should produce output as reaction to the clk-pulses.

That is correct if the GAL knows it is in the programming mode - otherwise it processes all inputs as regular inputs and applies previously programmed logic/algorithm to outputs. So you need to check whether the VPP_EN (Arduino IO 11) is enabled during programming and if so, whether the Edit pin (GAL pin 2) has 12V. Ensure the power switch on Afterburner is ON (LED is lit)

> Also, All the VIL-Pins should be tied to ground in edit-mode.

Correct, unless the GAL forces them high via internal algorithm.

> It seems on the board they are not connected to the Arduino at all

Disconnect Arduino (USB and voltage source), unplug the Afterburner board from Arduino, unplug the GAL from the socket. Verify the resistance between each of the pins (13 -17) on the Afterburner connector and ground pin (so check 13-GND, 14-GND etc.). Each of these pins should read around 4700 Ohm on the multimeter.

mr-who2012, 2022-12-15:
Thank you. I will try all that and report back. I opened a diskussion [here](https://github.com/ole00/afterburner/discussions/24) and posted a quick answer.