ole00/afterburner #27: GAL22V10D support?

sanlun777, 2023-04-20:
Only the D version is readily available where i am. Has it been tested? The docs only list the B version (GAL22V10B).
Haven't had a chance to test it myself yet, as i'm missing some IC and waiting for it to arrive, but seeing all the compatibility with something not supported issues, thought it would be nice to ask first.

ole00, 2023-04-20:
Yes, the D version works. Just tried it and it erases and programs fine.
[image]

sanlun777, 2023-04-21:
Well, that was fast, this is fantastic news! Thanks so much for the fast response and for your time. Closing now

closed 2023-04-21 by sanlun777