ole00/afterburner #28: MCP4151 instead of MCP4131

MarkusPlehn, 2023-05-15:
is it possible, to change the MCP4131 with the 4151 because of availability?
It's hard to find these part here in germany. But the MCP4151 is available.

ole00, 2023-05-15:
Not as a drop-in replacement (that is, without Arduino code changes) . The reason it's not possible as a drop-in replacement is that 4151 has 257 tap values, while 4131 has only 129.  So setting the 4131 maximum tap of value 128 would only set the 4151 tap to a half of its resistance range.

On the other hand, it is possible to use 4151 if you also change the Arduino code and recalculate the tap value (multiply by 2) just before it is sent to the MCP chip via SPI transaction.

BTW. I just checked mouser.de and digikey.de and there seems to be plenty of stock of mcp4131.

ole00, 2023-05-15:
If you look at aftb_mcp4131.h source code, function  mcp4131_reg(...). The function parameters are 'address' and 'value'. What needs to be done is: If the address equals ADDR_WIPER then multiply the value by 2. do that on the start of the function. Also, when the 'read_reg' parameter is true then you have to divide the Wiper value read from the chip by 2 just before tha value is returned from the function (only when ADDR_WIPER address is specified).

Thinking about it, the code could autodetect mcp4131 and mcp4151 by setting value of the tap/wiper to 200. If the value would be accepted by the chip (read value would also be 200) then it is mcp4151. The value multiplication/division could then be done conditionally based on the detected chip.

MarkusPlehn, 2023-05-15:
thanks for your reply. yes Mouser and the other big player could deliver,
but with high shipping cost (single order).

The automatic chip detection inside the code is a quite nice idea 👍

ole00, 2023-05-16:
Chip availability: you have a good point. Even if the parts are in stock today, there is no guarantee they will be available in 6 months as well.
I've ordered mcp4151 and will add support for the chip. Thanks for the idea.

ole00, 2023-05-19:
I got he MCP4151 today, so I've added support for it.  The chip type (4131/4151) is autodetected, although slightly different technique had to be used. It seems only 'Autoincrement' commands clamps the value to its upper limit, so I used that to do the autodetection. In order to use MCP4151-103 make sure to pull the latest code and upload the Arduino sketch. The MCP4151 is now a hardware drop-in replacement  for MCP4131, no need to modify the board.
[image]

MarkusPlehn, 2023-05-20:
Great! i have finished my Board today and will test withe the MCP4151 👍

MarkusPlehn, 2023-05-20:
YES, it works. 😃
I have to calibrate with Option 'B5' then the measurement was exactly.

ole00, 2023-05-20:
Well done. I'm glad it works for you.

Skiner2, 2025-03-09:
Hi.
Is it possible to use MCP41010?

ole00, 2025-03-10:
MCP41010 seems to be pin compatible, but the software protocol that sets the pot values is different. So, it is not a drop-in replacement, the Arduino software that controls the pot would have to be rewritten to make it work. To answer your question: it should be possible to use if you want to spend some time modifying the Afterburner's software.