ole00/afterburner #31: Not all MT3608 modules are wired the same

meup, 2023-06-04:
I just put together an Afterburner, and I had a heck of time trying to get it calibrated.

I eventually figured out that the MT3608 module that I was using had the wiper and low-side terminals of the pot shorted together.
At first, I tried to work around this by lowering R2's value, but I wasn't able to achieve the 5V-16.5V range.
I eventually chose to desolder the module, desolder the pot, cut the shorting trace, and put it all back together again.
Eventually, I could get it calibrated.

Btw, if you change line 196 of aftb_vpp.h to have less-than-or-equal instead of just less-than, it will avoid an odd corner case for when the voltage reading fails to increase initially and it sets all the indexes to 1.  Of course, you normally shouldn't run into this unless you've got an odd MT3608 module.

Finally, I also faced the issue of my Arduino's A/D measurement of VPP being off (low) by 0.8V (for 16.5V).  I had to increase the value of R7 by 2.4K to get the readings to within 0.1V.  I'm not quite sure why that was the case.

Anyway, I got my chip programmed, and I'd like to thank you for putting this project together.

meup, 2023-06-04:
[image]

ole00, 2023-06-04:
Thanks for the report and congrats to making it to work! Also, thanks for the image, that's very helpful for others not to fall into the same trap as you. I was not aware such incompatible MT3608 modules exist. I desoldered some of my broken modules (which I damaged during my trials and errors) for comparison. It looks like there is an indication how to spot the incompatible module - I'll put the image with a warning on the main page.

[image]

ole00, 2023-06-04:
@meup : could you share where you bought your mt3608 module and possibly the seller's name? Thank you.

meup, 2023-06-04:
I got it on Ebay from seller [koo97ie09].  However, their pictures showed several different versions, so which are shipped is probably just the luck of the draw.  Testing for the trace is easy: measure the resistance between the wiper terminal and the lower side of the 2.2K resistor - it will be continuous if the trace is there.  Removing the pot wasn't difficult either: my soldering iron could heat the two wide terminals at the same time, so I just had to alternate heat between the two and the one terminal while using a pointy tweezer stuck under it to provide pressure to lift it away from the board.  I used the same tweezer to clean out the holes, since I didn't have a solder sucker or braid handy.  I used a sharp knife to cut where your purple arrow points (and measured to verify).

The component footprint under the POT is for attaching a fixed resistor instead of the POT for fixed voltage output.

Thinking about this a bit, I suppose an alternate approach to desoldering the POT and cutting the trace is to replace the 2.2K resistor with a ~27K one.  But SMD rework might not be so easy either.

hubertushirsch, 2024-01-27:
I purchased MT3608 modules for Afterburner.
These modules have all the characteristics of the "good modules":
- Wiper screw on the right side
- VIN/VOUT text in sans font
- no component under the pot
Nevertheless, the afterburner could not be calibrated.
After removing the pot, I saw that the wiper was connected to the terminal leading to the 2k2 resistor.
After disconnecting the two connections, the MT3608 module worked perfectly.
[image]
Greetings
Hubert

ole00, 2024-01-28:
@hubertushirsch thanks for the report. I have one of those and I considered it a good module - clearly I was wrong. I rechecked my module (that looks the same as on your photo) and indeed it has the same connection traces you marked for cut. Perhaps there is a way how to make those modules to work without desoldering the pot (replacing the 2k2 resistor with a higher value?).

ole00, 2024-01-28:
I did a test with the "bad" type of the mt3608 module. They can be used without desoldering the potentiometer by replacing the 2k2 resistor by 15k (possibly 18k) resistor. Using 12k or 20k gives sub-optimal results. Using 10k gives a wrong calibration result.  See the the #47 thread.