ole00/afterburner #33: Change serial speed.

TheEccentricGenius, 2023-08-30:
So I've tried 57600, and it's pretty hard to get it to work reliably.

I would prefer 115200, only my MCP2200 serial chips work at 57600.
I think it would be easier to use 115200, all UNO boards have to be accurate enough to upload at that speed.

Any reasons for not going with 115200(So I'm not left in the dark)?

ole00, 2023-08-30:
Not quite sure why your MCP2200 works only at 57600 - do you mean in general with any software or specifically with Afterburner? If you mean with Afterburner, then the baud rate is hardcoded in the UNO sketch and the pc program - both must have the same baud rate in order to have serial connection.  Baud rate of 115200 seems to be worse on UNO (16MHz crystal clock) with error rate -3.5%, while 57600 has error rate 2.1%. The best baud rates on Arduino UNO seem to be 9600 or 38400. Try them out - it might work better for you. You'll need to change and upload  the Afterburner sketch ( check for Serial.begin(57600)) and recompile PC program (change baud rate in src_pc/serial_port.h - the code there is specific for Windows and Linux, so to be sure change all occurrences of 57600 to your desired baud rate).

TheEccentricGenius, 2023-08-31:
Sorry, only the MCP2200 chips work with the 57600 baud rate....And they still a bit glitchy.

I am also using the 8MHz internal clock....which is yet worse for 57600 baud.
I can't explain the 115200 baud rate....all my ATmega328P's run with that baud rate without a problem.

I ended up changing the baud rate to 115200 which is rock solid reliability.
Thanks for the info....Sorry if I seemed a little "off" the other day, I was really  frustrated by it all.