ole00/afterburner #41: PC program does not recognize output sometimes

rhgndf, 2023-10-14:
Errors with:
```
Afterburner v.0.5.3 
opening serial: /dev/ttyACM0
Output from programmer not recognised: 
AFTerburner v.0.5.3
 varVpp 
type 'h' for help
>
result=-1
```
Possibly due to the function openSerial in afterburner.c: `buf[total - 3] == '>'` where there is variability on the newlines read at the end, so might need to check for '>' in the last few bytes as compared to only checking one position?

ole00, 2023-10-14:
Thanks for the report and the insight. So far I can't reproduce it, but I admit it can happen if the '>' character is received in a separate 'batch' of characters read from the serial line. In such case the function checkPromptExists() returns earlier than it should and causes the issue.

ole00, 2023-10-14:
This might be improved (fixed?) by this commit b498bc7

rhgndf, 2023-10-15:
Seems to be more reliable now

closed 2023-10-15 by rhgndf