ole00/afterburner #44: Handling >= 10 GAL types

rhgndf, 2023-11-01:
In afterburner.ino line 2252 the type is decoded as `char type = line[1] - '0';`, but in the pc program it is sent as `sprintf(buf, "#t %i\r", (int) gal);`. Above 10, the value will be two chars, which makes the decode wrong.

This can be solved with one of these two solutions:

1. Make the arduino side parse an integer
2. Send `sprintf(buf, "#t %c\r", gal + '0');` instead

ole00, 2023-11-01:
Thanks for the reminder. I've noticed that as well and fixed it locally when I tested/merged your new branch (when the total number of gals increased past 9). I used method 2 in function operationSetGalType() and operationReadFuses(). I'm still waiting for my new GAL18V10, once I verify the code works I'll merge it. Sorry for the delay.

rhgndf, 2023-11-02:
I didn't notice this until I tried to read the PES from a ATF750 that I received recently. Wanted some input on which one is a better solution to fix this issue in the code.

Unfortunately it seems like the ATF750 is broken. It gets really hot after a while in the socket and I can't read the PES.

ole00, 2023-11-04:
I've pushed the fix for the gal type > 9 (I used method 2).

ATF750 - yes, it might be quite complex to make it work.  GALS get hot if they are exposed to high VPP for long amount of time (even more than 3 seconds might be too much). Normally the VPP is On only for less than a second. VPP higher than expected might also fry the GAL chip inside. ATFs usually work with VPP ~ 10V - 10.5V. VPP 12V might be too much for ATF750.

rhgndf, 2023-11-04:
There's quite a few more lines, like in line 1109.

ole00, 2023-11-04:
Ah.. , you are right. Thanks for that - I'll fix it.

rhgndf, 2023-11-05:
Also is there a reason you removed strncmp_P? It was using PROGMEM, so it doesn't consume RAM and it makes the code is a bit more readable?

rhgndf, 2023-11-05:
Sorry to tag on this on the issue but I have a GAL18V10B-15LP and a GAL18V10-15LP and both read and write fine, so I think you might have gotten a bad batch or something. Both have printed markings instead of laser etched.

ole00, 2023-11-05:
I replaced strcmp_P beacuse  to me it is less readable, it introduces dependency to another function/library and the code is ~40 bytes bigger. It's fine to use stncmp on PC app though as the size of the PC app does not matter. I prefer MCU code to be as small as possible.

GAL18V10B-15LP : it's good to know it works for you. I did not write none of them work - I wrote in Troubleshooting section that **some** may not work because of being fake - I think that corresponds to what you described: " I think you might have gotten a bad batch or something. Both have printed markings instead of laser etched.".

> I doubt you need to remove the resistor for version 3.0 mod, since it just acts as a pull-down and a 5V will do just fine since the resistance is big enough.

You might be right, though it looks like the modified board works OK without the resistor. Would there be benefit having the resistor (genuine question)?

rhgndf, 2023-11-05:
For the GAL18V10s, I just wanted to share how my GALs looked, wasn't intending to point to the troubleshooting section.

Fair point on the code size though.

I think there might be a case for pull-downs on all the pins since it avoids floating inputs. Or a resistor between the GAL and the rest of the circuit when it is operating in non-programming mode and the outputs are being driven. On the same note, there could also be a current limiting resistor on the connection between the output of MT3608 and the GAL.

ole00, 2023-11-05:
> I think there might be a case for pull-downs on all the pins since it avoids floating inputs.
I agree, but specifically on ZIF20 the pin is either HIGH or LOW (but not Z state) because it is connected to the shift register's output pin. If the pin is floating then it is meant to be floating (after read/write operation finished). 

> On the same note, there could also be a current limiting resistor on the connection between the output of MT3608 and the GAL.
I'm not sure how this VPP is utilised on different GALs. Adding a limiting resistor in series to VPP might break the fuse writing functionality. I guess the different VPP options (10-16.5V) are not there just as some sort of signalling or protection from accidental erase. I think (not confirmed) that the VPP is used physically to support flash writing, therefore limiting the current might cause fuse writing issues. Also, it probably depends on the size of the resistor - I you meant a small resistor (10-50 Ohms) then it would probably work OK. OTOH it would be good to know how much current the VPP draws.

rhgndf, 2023-11-05:
The VPP is definitely used for erasing. Modern devices feature a charge pump that generates the appropriate high voltage in the chip itself to do the erase, see the section on EEPROMs in https://en.wikipedia.org/wiki/Charge_pump.

closed 2023-11-07 by rhgndf