ole00/afterburner #47: Unable to calibrate VPP

AJROMR, 2024-01-26:
I have built v3.1a pcb using SMT components, a good MT3608 module but I am still having issues with afterburner calibration.

[image]

In order to get detailed info I have made the following modifications to the sketch files:

---
afterburner.ino (Line# 2703):
if (offset >=0 && offset <= 9) {

aftb_vpp.h (Line# 41)
#define VPP_VERBOSE 1

---

I have tested it with two different Arduino UNO R3 (chinese clones) and as I was not able to find the issue I have built the following circuit in a breadboard but the results were very similar.

[image]

There is only one change, in the v3.1a pcb I used a MCP4131-103 digital pot, in this breadboard version I used a DIP-8 MCP4151-103 digital pot.

v3.1a pcb (SMT) calibration output:

---
./afterburner s

Turn the Pot on the MT3608 module to set the VPP to 16.5V (+/- 0.05V)

varSetVppIndex 64
varSetVppIndex 112
varSetVppIndex 124
varSetVppIndex 126
varSetVppIndex 128
16.52
16.52
16.52
16.52
VPP OK
varSetVppIndex 0

---
./afterburner b

Using cal offset: 25
VPP voltages are scanned - this might take a while...

VPP calib. offset: 25
varSetVppIndex 1
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 900 is 1
vppIndex 1
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 950 is 1
vppIndex 2
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1000 is 1
vppIndex 3
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1050 is 1
vppIndex 4
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1100 is 1
vppIndex 5
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1150 is 1
vppIndex 6
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1200 is 1
vppIndex 7
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1250 is 1
vppIndex 8
varSetVppIndex 1
1) varSetVppIndex 2
2) *Index for VPP 1300 is 1
vppIndex 9
varSetVppIndex 1
1) varSetVppIndex 2
2) varSetVppIndex 3
3) varSetVppIndex 4
4) varSetVppIndex 5
5) *Index for VPP 1350 is 4
vppIndex 10
varSetVppIndex 4
4) varSetVppIndex 5
5) *Index for VPP 1400 is 4
vppIndex 11
varSetVppIndex 4
4) varSetVppIndex 5

---
./afterburner m

VPP calib. offset: 25
VPP: 4.2 - 5.0V : varSetVppIndex 0
11.25
VPP: 9.0V : varSetVpp 0:1
varSetVppIndex 1
12.95
VPP: 12.0V : varSetVpp 6:1
varSetVppIndex 1
12.94
VPP: 14.0V : varSetVpp 10:4
varSetVppIndex 0
varSetVppIndex 4
12.97
VPP: 16.0V : varSetVpp 14:121
varSetVppIndex 58
varSetVppIndex 82
varSetVppIndex 94
varSetVppIndex 100
varSetVppIndex 103
varSetVppIndex 104
varSetVppIndex 105
varSetVppIndex 106
varSetVppIndex 107
varSetVppIndex 108
varSetVppIndex 109
varSetVppIndex 110
varSetVppIndex 111
varSetVppIndex 112
varSetVppIndex 113
varSetVppIndex 114
varSetVppIndex 115
varSetVppIndex 116
varSetVppIndex 117
varSetVppIndex 118
varSetVppIndex 119
varSetVppIndex 120
varSetVppIndex 121
15.98

---

Breadboard calibration output:

---
./afterburner s

Turn the Pot on the MT3608 module to set the VPP to 16.5V (+/- 0.05V)

varSetVppIndex 64
varSetVppIndex 112
varSetVppIndex 124
varSetVppIndex 126
varSetVppIndex 128
16.48
16.50
16.52
16.49
VPP OK
varSetVppIndex 0

---
./afterburner b

Using cal offset: 25
VPP voltages are scanned - this might take a while...

VPP calib. offset: 25
varSetVppIndex 1
varSetVppIndex 1
1) varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 900 is 2
vppIndex 1
varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 950 is 2
vppIndex 2
varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 1000 is 2
vppIndex 3
varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 1050 is 2
vppIndex 4
varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 1100 is 2
vppIndex 5
varSetVppIndex 2
2) varSetVppIndex 3
3) *Index for VPP 1150 is 2
vppIndex 6
varSetVppIndex 2
2) varSetVppIndex 3
3) varSetVppIndex 4
4) *Index for VPP 1200 is 3
vppIndex 7
varSetVppIndex 3
3) varSetVppIndex 4
4) varSetVppIndex 5
5) *Index for VPP 1250 is 4
vppIndex 8
varSetVppIndex 4
4) varSetVppIndex 5
5) *Index for VPP 1300 is 4
vppIndex 9
varSetVppIndex 4
4) varSetVppIndex 5
5) *Index for VPP 1350 is 4
vppIndex 10
varSetVppIndex 4
4) varSetVppIndex 5
5) varSetVppIndex 6
6) *Index for VPP 1400 is 5
vppIndex 11

---
./afterburner m

VPP calib. offset: 25
VPP: 4.2 - 5.0V : varSetVppIndex 0
11.39
VPP: 9.0V : varSetVpp 0:2
varSetVppIndex 2
13.04
VPP: 12.0V : varSetVpp 6:3
varSetVppIndex 3
13.04
VPP: 14.0V : varSetVpp 10:5
varSetVppIndex 0
varSetVppIndex 5
13.06
VPP: 16.0V : varSetVpp 14:10
varSetVppIndex 3
varSetVppIndex 10
13.11

---

On the other hand,
1) README.md: There are several mentions to R8 but I think you want to mean R9.
2) bom.txt: "(R1 - 5x)" but in the pcb there are only 4 footprints for R1.

Thanks in advance

ole00, 2024-01-27:
Hi,

sorry to read about the calibration issue.
few observations:
1) the MT3608 does look OK to me - like the type that should work fine
2) Arduino UNO Chinese clones - they usually work OK, I find them slightly better that genuine Arduino UNO
3) R8 /R9 mismatch and BOM - yes, on recent PCB it is R9 POT. I'll fix the description, thanks.
4) MCP4131 and MCP4151 is autodetected in Arduino software, both should work OK.
5) when using './afterburner s' you did not mention whether you measured the output of the MT3608 module by a multimeter. It is recommended, to verify the reading from Arduino matches real value.
6) when using './afterburner b' I can see the issue : your listing prints '*Index for VPP 900 is 1'.   The value must be between 15 and 35. Value 1 is definitely a wrong result. That either means the R9 pot is set to an incorrect extreme, or the Arduino ADC reading VPP_SENSE signal reads a wrong value. Double check that the R5 (100k) and R6 (20k) on the voltage divider are the correct values on your board.  Also, double check that R2 is 3k3 on your board.  

So, ensure the multimeter reading (point 5) matches the print out from Arduino. Then double check the resistors (point 6)
and if it is OK then rotate R9 pot to extremes and repeat './afterburner s' and './afterburner b'. If you rotate Pot R9
you must re-do the './fterburner s' and make sure the voltage output is 16.5V (also on multimeter). Only then you do './afterburner s'.  Check the listing '*Index for VPP 900 is [X]'in both cases (R9 extremes). The [X] should really be different for both extremes. 
Let me know how it goes.

ole00, 2024-01-27:
Just for a reference, here is my printout when running  ./afterburner b
ER: cal offset failed
VPP voltages are scanned - this might take a while...

VPP calib. offset: 15
*Index for VPP 900 is 15
*Index for VPP 950 is 34
*Index for VPP 1000 is 49
*Index for VPP 1050 is 61
*Index for VPP 1100 is 72
*Index for VPP 1150 is 81
*Index for VPP 1200 is 88
*Index for VPP 1250 is 95
*Index for VPP 1300 is 101
*Index for VPP 1350 is 106
*Index for VPP 1400 is 111
*Index for VPP 1450 is 115
*Index for VPP 1500 is 118
*Index for VPP 1550 is 122
*Index for VPP 1600 is 125
*Index for VPP 1650 is 128
Calibration OK

ole00, 2024-01-27:
I've printed the calibration when Pot R9 is in the extremes. It looks quite different from what you sent earlier:

Non optimal calibration: R9 rotated fully counter-clockwise:
VPP calib. offset: 15
*Index for VPP 900 is 1
*Index for VPP 950 is 1
*Index for VPP 1000 is 17
*Index for VPP 1050 is 37
*Index for VPP 1100 is 52
*Index for VPP 1150 is 65
*Index for VPP 1200 is 76
*Index for VPP 1250 is 86
*Index for VPP 1300 is 94
*Index for VPP 1350 is 101
*Index for VPP 1400 is 107
*Index for VPP 1450 is 112
*Index for VPP 1500 is 117
*Index for VPP 1550 is 121
*Index for VPP 1600 is 125
*Index for VPP 1650 is 128
Calibration OK

Non optimal calibration: R9 rotated fully clockwise:
VPP voltages are scanned - this might take a while...

VPP calib. offset: 15
*Index for VPP 900 is 73
*Index for VPP 950 is 80
*Index for VPP 1000 is 87
*Index for VPP 1050 is 92
*Index for VPP 1100 is 97
*Index for VPP 1150 is 102
*Index for VPP 1200 is 106
*Index for VPP 1250 is 109
*Index for VPP 1300 is 112
*Index for VPP 1350 is 115
*Index for VPP 1400 is 118
*Index for VPP 1450 is 120
*Index for VPP 1500 is 122
*Index for VPP 1550 is 124
*Index for VPP 1600 is 126
*Index for VPP 1650 is 128
Calibration OK

ole00, 2024-01-27:
And perhaps a little explanation of what the Index [x] means. The Digi POT has 128 positions (or taps) and each position provides a different resistance. Each POT position (or resistance) gives a different output voltage on the MT3608 module. When running the 'b' command, the Arduino sets the POT to every position and records which position corresponds to each output voltage (VPP, which is sensed by Arduino's ADC) that is used to program the GAL chips. Of course, to make it working correctly the Arduino's ADC must read the VPP voltage right - that's why the multimeter verification is required.

AJROMR, 2024-01-28:
Thank you very much for your reply.

> 1. the MT3608 does look OK to me - like the type that should work fine

I have 3 or 4 different types, all of them should be working according to your pics.

> 4. MCP4131 and MCP4151 is autodetected in Arduino software, both should work OK.

I used MCP4131 in SMT v3.1a board version and MCP4151 in the breadboard version (VPP circuit) but I am getting same results with both boards.

[image]
[image]
[image]

> 5. when using './afterburner s' you did not mention whether you measured the output of the MT3608 module by a multimeter. It is recommended, to verify the reading from Arduino matches real value.

Sorry, './afterburner s' measures "16.50" but the multimeter ~16.16VDC.

> 6. when using './afterburner b' I can see the issue : your listing prints '*Index for VPP 900 is 1'. The value must be between 15 and 35. Value 1 is definitely a wrong result.

I know but I am just getting 1.

> That either means the R9 pot is set to an incorrect extreme

R9 is set exactly to 1kΩ (between pins 1 & 2).

> Double check that the R5 (100k) and R6 (20k) on the voltage divider are the correct values on your board.

I have also measured/checked both resistor values using the TH drills.

[image]

(sorry about the flux, I have checked/reworked all the solderings several times)

> Also, double check that R2 is 3k3 on your board.

The ohmeter measures 3301Ω between R2 pin (S) and R9 pin 2.

[image]

---
./afterburner s
Turn the Pot on the MT3608 module to set the VPP to 16.5V (+/- 0.05V)

16.50
16.50
16.51
16.50
VPP OK

---
./afterburner b
ER: cal offset failed
VPP voltages are scanned - this might take a while...

VPP calib. offset: 25
*Index for VPP 900 is 1
*Index for VPP 950 is 1
*Index for VPP 1000 is 1
*Index for VPP 1050 is 1
*Index for VPP 1100 is 1
*Index for VPP 1150 is 1
*Index for VPP 1200 is 1
*Index for VPP 1250 is 1
*Index for VPP 1300 is 10
*Index for VPP 1350 is 50
*Index for VPP 1400 is 75
*Index for VPP 1450 is 92
*Index for VPP 1500 is 105
*Index for VPP 1550 is 115
*Index for VPP 1600 is 123
*Index for VPP 1650 is 128
Calibration OK

---

I guess there is something wrong but I cannot figure what is happening. I'd appreciate any help. Thanks in advance.

ole00, 2024-01-28:
Hi,

it looks like the issue is mt3608 module. I thought this type of the module is the older (good/working) type, but it looks like it is not. Just yesterday, another user @huberthirsch reported the issue (#31 , bottom of the thread) with that type of the module. Apology for that, that was my mistake.

I think it might be possible to fix the mt3608 module without desoldering the potentiometer.  I think a replacement of the 2k2 resistor (in between the pot and the capacitor) with a higher value should do the trick (perhaps a value like 20k or 30k), but I'd have to investigate. 
BTW. I like the way you connected the module  by using the headers.

ole00, 2024-01-28:
I did a few tests. The new modules can be used be used without desoldering the pot by replacing the 2k2 resistor by 12k or 15k (possibly 18k) resistor. It is a simple mod: heat the soldering iron to 350 deg C, put a medium size solder blob on the tip of the soldering iron, touch the resistor (in the middle) for a second (it heats up and the solder pads get melted), then slide it off from the board by applying a slight pressure on the resistor.  Soldering the new resistor is not hard, I tried a through hole resistor (when testing 20k resistor - did not give me a good result), 10k SMT resistor (wrong results)  and then 12k SMT resistor (0605 package) which works OK (not ideal, but should do).  I did not test 15K or 18k replacement resistors (I don't have them available now), but I'd reckon these should be fine.

The 12k resistor gives the following calibration values (with the new type of the mt3608 module):
*Index for VPP 900 is 11
*Index for VPP 950 is 32
*Index for VPP 1000 is 47
*Index for VPP 1050 is 60
*Index for VPP 1100 is 71
*Index for VPP 1150 is 80
*Index for VPP 1200 is 88
*Index for VPP 1250 is 95
*Index for VPP 1300 is 101
*Index for VPP 1350 is 106
*Index for VPP 1400 is 111
*Index for VPP 1450 is 115
*Index for VPP 1500 is 119
*Index for VPP 1550 is 122
*Index for VPP 1600 is 125
*Index for VPP 1650 is 128

The 12k resistor is the absolute minimum for Afterburner, when I tried 10k the 4.2 - 5V VPP gave me 5.66V which is out of spec. Do not use 10k resistor replacement. The 20k replacement should work, but the calibration values are less than ideal. I'd reckon the best option is to use 15k resistor replacement.
Hope it helps.

AJROMR, 2024-01-28:
> it looks like the issue is mt3608 module. I thought this type of the module is the older (good/working) type, but it looks like it is not. Just yesterday, another user @hubertushirsch reported the issue (#31 , bottom of the thread) with that type of the module. Apology for that, that was my mistake.

No problem at all. During the last 3 years I have bought several dozen of them and have not been able to find one that works properly with this pcb.

> BTW. I like the way you connected the module by using the headers.

Thank you. This makes it easier to test one MT3608 module after another. It would be better if the board had an extra FB pad just below the MT3608's FB pin.

> I did a few tests. The new modules can be used be used without desoldering the pot by replacing the 2k2 resistor by 12k or 15k (possibly 18k) resistor.

That did the trick!

[image]

---
./afterburner s
Turn the Pot on the MT3608 module to set the VPP to 16.5V (+/- 0.05V)

16.50
16.50
16.50
16.51
VPP OK

---
./afterburner b
ER: cal offset failed
VPP voltages are scanned - this might take a while...

VPP calib. offset: 25
*Index for VPP 900 is 25
*Index for VPP 950 is 38
*Index for VPP 1000 is 50
*Index for VPP 1050 is 59
*Index for VPP 1100 is 67
*Index for VPP 1150 is 74
*Index for VPP 1200 is 81
*Index for VPP 1250 is 86
*Index for VPP 1300 is 91
*Index for VPP 1350 is 95
*Index for VPP 1400 is 99
*Index for VPP 1450 is 103
*Index for VPP 1500 is 106
*Index for VPP 1550 is 109
*Index for VPP 1600 is 112
*Index for VPP 1650 is 114
Calibration OK

---
./afterburner m

VPP calib. offset: 25
VPP: 4.2 - 5.0V : 5.05
VPP: 9.0V : 9.00
VPP: 12.0V : 12.03
VPP: 14.0V : 13.97
VPP: 16.0V : 16.04

---

Thank you very much!

AJROMR, 2024-01-29:
Finally remind you, just in case, that in the file bom.txt where it says:

---
BOM for Afterburner 3.0
=======================
* (R1 - 5x) Resistor 4k7 (TH or SMT 0805) 0.25W, 5% or better tolerance  - 5x

---

it should read:
---
BOM for Afterburner 3.1a
========================
* (R1 - 4x) Resistor 4k7 (TH or SMT 0805) 0.25W, 5% or better tolerance  - 4x

---

ole00, 2024-01-29:
Thanks for the feedback - it's good to know 18k resistor works well. I'm glad it finally works for you.

> This makes it easier to test one MT3608 module after another. 

Definitely good for testing, but the drawback is the decoupling capacitors are further apart from the booster convertor so they are not as effective filtering the ripples.

> It would be better if the board had an extra FB pad just below the MT3608's FB pin.

OK. that can be done - I guess the only reason would be to hide the connection wire? I placed the FB pad away from the MT3608 area to be able to conveniently check the pin and also to make the connection point obvious on photographs for the documentation purposes.  The whole confusion about the mt3608 modules (which type works without mods and which type needs a mod) is a bit unfortunate, so I'll probably redesign the area underneath the mt3608 modules and put there extra footprints so that the voltage booster can also be built from discrete parts as an option.

> Finally remind you, just in case, that in the file bom.txt where it says...

Thanks, previously there were 5 resistors, one was removed recently. There is a chance I'll put the 5th one back (I'll need to do more tests) - it seems it's not critical to have the 5th one there though. I'll update the BOM in the meantime.

ole00, 2024-01-29:
One more thing. I just noticed your calibration listing. This particular line is unusual:

*Index for VPP 1650 is 114

Usually, when (during the 's' command) the reference voltage is set to 16.5V then the  last index (for VPP 16.5V) should be 128 (not 114 as you have in your listing). It is not a big issue, just an observation (the code and voltages should still work fine). It would be better to retest the VPP values with multimeter while the 'm' command is running to be absolutely sure the VPP voltages are correct.

AJROMR, 2024-01-29:
> it's good to know 18k resistor works well. I'm glad it finally works for you.

Thank you again. 

> Definitely good for testing, but the drawback is the decoupling capacitors are further apart from the booster convertor so they are not as effective filtering the ripples.

Right.

> OK. that can be done - I guess the only reason would be to hide the connection wire?

I was talking an extra pad keeping the previous one. In this way if I solder a pin to the pot middle pin there is no need for the wire (it also allows to solder a pin socket).

> I'll probably redesign the area underneath the mt3608 modules and put there extra footprints so that the voltage booster can also be built from discrete parts as an option.

That would be great. 

> I think this issue/solution might be helpful to others, so I'll keep it opened for others as a reference. I usually do not close issues for that reason, it easier to find answers to the same/similar questions.

That is fine. I did it because this site encourages closing resolved issues.

> One more thing. I just noticed your calibration listing. This particular line is unusual:

I will check it out. Thanks.

AJROMR, 2024-01-30:
> *Index for VPP 1650 is 114

I was tweaking the pots and I have finally good VPP values (measured with the multimeter).

Now I get:
*Index for VPP 1650 is 124

using "./afterburner b" VPP is in the range from 8.3V (9.0V) to 16.52V (16.5V) (measured with the oscilloscope),

[image]
(DC offset is -12.00V)

and I get much better results with "./afterburner m". Almost all the values read are exactly as expected and the few that do not match exactly, have a maximum deviation of 0.03V.

ole00, 2024-01-31:
OK. looks good. Deviation of - 0.1V to +0.05V  should be OK, most of the GAL chips will work fine. And if not, the GAL chip will not be fried/bricked. If in doubt, then  a small undershoot (0.05V) might be the best because the voltage booster IC has a tendency to overshoot when the voltage is raised. This overshoot is partially compensated by the filtering cap on the output and by the software which raises the voltage gradually to the target voltage value.

ole00, 2024-01-31:
Also remember, when you swap the Afterburner (ie. the one made on PCB and the one made on breadboard) you should recalibrate. That's because the resistors for sensing (voltage dividers), 3V VREF have +/-1% deviation and it might is some cases be enough to cause slightly different voltages being produced. This also applies if you use and/or don't use external power supply. The 5V rail might change voltages significantly (in those 2 cases) and the voltage booster IC will produce slightly different outputs. As long as you don't change the way the Afterburner is powered (same USB cable, same USB port on your PC etc.) the calibrated output voltage should stay valid.

AJROMR, 2024-02-01:
Thank you again.

Despite the good results with the calibration and having managed to get:

"*Index for VPP 1650 is 128"

there was no such luck attempting to program two different PLDs, ATF22V10C and GAL22V10B. afterburner reads the JED file correctly, but when it tries to burn it stops and shows error on 4858 fuses. It is always the same result despite the PLD or VPP calibration.

I am currently waiting for a new batch of MT3608 DC-DC converters (HW-045 type) which should be "good modules". Hopefully I will have better luck with them.

ole00, 2024-02-02:
That's interesting. Is the ATF22V10C brand new? If you write the fuses (with error) and then read the fuses back (r command) - can you at least see that some fuses were burned? Did you also erased the fuses with 'e' command? Could you try to burn the following conditioning JED file (it is literally just 2 lines):

```
*F0
*

```

Then if you erase and write with errors - could you write (next time without erasing) again several times? Will the amount of failed fuses become smaller the more write attempts you do?

AJROMR, 2024-02-04:
> Is the ATF22V10C brand new?

Yes, it is.

> Could you try to burn the following conditioning JED file (it is literally just 2 lines):
> *F0
> *
> then if you erase and write with errors - could you write (next time without erasing) again several times? Will the amount of failed fuses become smaller the more write attempts you do?

---
afterburner s
Turn the Pot on the MT3608 module to set the VPP to 16.5V (+/- 0.05V)

16.50
16.50
16.50
16.50
VPP OK

---
afterburner b
Using cal offset: 0
VPP voltages are scanned - this might take a while...

VPP calib. offset: 0
*Index for VPP 900 is 35
*Index for VPP 950 is 49
*Index for VPP 1000 is 60
*Index for VPP 1050 is 70
*Index for VPP 1100 is 78
*Index for VPP 1150 is 86
*Index for VPP 1200 is 92
*Index for VPP 1250 is 98
*Index for VPP 1300 is 103
*Index for VPP 1350 is 107
*Index for VPP 1400 is 111
*Index for VPP 1450 is 115
*Index for VPP 1500 is 118
*Index for VPP 1550 is 121
*Index for VPP 1600 is 124
*Index for VPP 1650 is 127
Calibration OK

---
afterburner m

VPP calib. offset: 0
VPP: 4.2 - 5.0V : 4.81
VPP: 9.0V : 8.99
VPP: 10.0V : 9.91
VPP: 12.0V : 12.01
VPP: 14.0V : 13.98
VPP: 16.0V : 15.99

---
afterburner e -t ATF22V10C

afterburner wv -t ATF22V10C -f test.jed
Uploading fuse map...
5892/5892 |########################################|
ER verify failed. Bit errors: 5

afterburner wv -t ATF22V10C -f test.jed
Uploading fuse map...
5892/5892 |########################################|
ER verify failed. Bit errors: 2

afterburner wv -t ATF22V10C -f test.jed
Uploading fuse map...
5892/5892 |########################################|
ER verify failed. Bit errors: 4

afterburner r -t ATF22V10C
JEDEC file for ATF22V10C
*QP24*QF5892*QV0*F0*G0*X0*
L0000 00000000000001000000000000000000000001000000*
L5808 00000000000000000000*
N UES 00 00 00 00 00 00 00 00*
L5828 0000000000000000000000000000000000000000000000000000000000000000*
N PES 33 43 30 31 56 32 32 46 31 FF*
C0040
*

ole00, 2024-02-04:
Could you list the contents of test.jed file? I'll try to upload it to my ATF22V10C.

ole00, 2024-02-04:
Also, uncomment #DEBUG_VERIFY in Arduino.ino and upload the sketch to your Arduino. It will print more info during verification, please send the log.

ole00, 2024-02-04:
Other than that, possibly one of those:
- verify the solder joints on the ground connection on the through-hole pin headers (Zif socket)
- solder an extra non-SMT 10uF electrolytic cap directly on the output of the MT3608 module (where the VOUT+ and VOUT- pads are)
- the ATF22V10C VPP is 10V, your seems a bit on the low side (9.91). Possibly recalibrate so that the 10V is a bit above (like 10.05), or  leave it as it is (10V measures as 9.91) and change afterburner.ino line 840 from VPP_10V0 to VPP_10V5 (I tried the VPP 10.5V on my ATF22V10C and it works OK, so it should be safe). Do do not go above 10.5V with VPP on Atmel GALs, it may damage them.  BTW. I undervolted my 10V tap to 9.87V and I got verification errors as well.

ole00, 2024-02-04:
For the sake of research, I tested my ATF22V10C, ATF16V8B and ATF16V8C with VPP 10.5 and also VPP 11V and they were fine (programming and verification). It might be helpful if somebody could check what VPP voltage other GAL programmers set for Atmel GALs. GALmate seems to set VPP 12V, so perhaps Afterburner could use VPP 11V - but that would needs more research. Perhaps currently used VPP 10.0V is on the very low edge of what Atmel GALs can operate and VPP 10.5V (or even 11V) would be more compatible.

ole00, 2024-02-05:
Please pull the latest changes from git, master branch. I have updated Afterburner.ino sketch to improve reliability of fuse writing. Hopefully that should fix the write and verification errors for you. If you've already experimented with the code changes I suggested above, please undo them and use the latest master. Let me know whether that fixed the issue. Thanks.

AJROMR, 2024-02-08:
Sorry for the delay in my reply, I am still waiting for the good old MT3608 modules and was convinced they would be delivered sooner but apparently it will take a few more days.

---
test.jed
*F0
*
---
afterburner m

VPP calib. offset: 0
VPP: 4.2 - 5.0V : 4.86
VPP: 9.0V : 9.01
VPP: 10.0V : 10.02
VPP: 12.0V : 12.01
VPP: 14.0V : 14.02
VPP: 16.0V : 16.03

---
afterburner wv -t ATF22V10C -f test.jed
Uploading fuse map...
5892/5892 |########################################|
ER verify failed. Bit errors: 5655

---
afterburner wv -t ATF22V10C -f test.jed
Uploading fuse map...
5892/5892 |########################################|
ER verify failed. Bit errors: 5655

ole00, 2024-02-09:
No worries about delays, take your own pace.

I'm a bit puzzled. It looks like the new change in the Afterburner's code and VPP recalibration made it even worse - no fuse was written.

It looks like you have an oscilloscope - could you check the VPP rail and 5V rail during 'w' command while your ATF2210C chip is in the ZIF socket?

ZIF pins: 2=VPP, 12=GND, 24=5V

There might be a small dip on the 5V rail when the VPP is turned on - that's expected. You should see a trace like this:

[image]

AJROMR, 2024-02-14:
> No worries about delays, take your own pace.

Thanks.

No luck with the new MT3608 modules, the seller has sent some that are almost identical to the ones I already had. I have returned them but had a chance to take a picture first.

[image]

> I'm a bit puzzled. It looks like the new change in the Afterburner's code and VPP recalibration made it even worse - no fuse was written.

I have no doubt about it, the results have been worse.
 
> It looks like you have an oscilloscope - could you check the VPP rail and 5V rail during 'w' command while your ATF22V10C chip is in the ZIF socket?
> 
> ZIF pins: 2=VPP, 12=GND, 24=5V

No problem, I will do it and in a few days I will share the results.

ole00, 2024-02-14:
I've built another Afterburner PCB, this one has the raised MT3608 module - the same way you have done it. I also used the new type of the MT3608 module and modded it by using 15k feedback resistor. So this Afterburner build should be very similar to yours (at least VPP-wise.) I found no issue writing to ATF22V10C by using this Afterburner board, so unless the MT3608 module is broken (or is not calibrated) it should not be the root of the problem you are seeing.