ole00/afterburner #52: Alternative do MCP413X .,

fredcwbr, 2024-02-12:
Howdy,
First of all, thanks for your great work. 

I'm trying to build the afterburner, but sourcing those MCP413X pots is really a problem. The pot alternatives are nos suited at all.

Was thinking in revamping the V3.0 board, using LM317 controlled by a PWM. It can withstand a 3.5Amp, and with a rebuild of the input to acomodate a much higher input voltage (say 30V), how much current would programming request? .. Do this 3A fit the bill ?

Another question is about the various voltages. In the code you use indexes to address wiper positions, considering a predefined adjustment, thats really neat. In this new proposed approach, a function handling this indexes would convert to the needed PWM , that way it would give out the correct ratio to LM317. 
And the final question. Found voltages ranging from 5V ~ 19V ., .. would there be more? .. .and what would be the minimum step requirement (in volts/step) ?

Best Regards..

ole00, 2024-02-12:
Hi Frederico,
thank you for your interest.

I'm sorry to read the MCP hardware is not available where you live.

Unless you want to go for a long journey or redesigning Afterburner's software and hardware I would discourage you from using alternative VPP (voltage programming) solution. Some users can't make the Afterburner to work with the original hardware, but I don't blame them - even a small oversight or imperfection during Afterburner's production can make it unusable. So even though Afterburner is a simple project there are certain complexities.

You have several options and it depends what your goals with Afterburner are:
1) Do you just want to program few GAL16V8 or GAL22V10 (or their Atmel equivalent) IC for another hobby project ?
  If so, then use the legacy hardware design (see version_4_legacy git branch) and set the VPP manually by turning the pot. The current software is compatible with the old design.

2) Do you need to program some other GALs (other than GAL16V8 or GAL22V10) for another hobby project?
  If so, use the legacy hardware design (see point 1) and design an adapter - it is very simple to design an adapter  and it is usually quite cheap. The adapter usually requires signal re-routing to different pins and possibly  few pull down resistors.

3) If you want it for an educational purpose, in a electronic club or hacker space?
  If so, use the legacy hardware design (see point 1) and chose one specific GAL chip you want to use. Then manually adjust VPP for that specific GAL chip and hot-glue the pot so that the students can't accidentally tamper with it.

4) If you really need programmable VPP, then buy MCP4151 which also works with Afterburner. The chip is available on   ebay and Aliexpress (and also on Digikey, Mouser, RS components etc.). It is not cheap, but will worth the money as it will save you a lot of effort redesigning the software and hardware.
  
5) If you really-really need programmable VPP and even MCP4151 is not a solution for you, then find another variable pot IC  that is available to where you live. A brief look at AD8400ARZ10 shows it might be simple to design an intermediate board that would fit into the MCP4131 footprint and it should behave similarly. A small software modifications would be required though.

6) If you really would like to use the solution you suggested, here are the issues I can see:

  * the PWM pin on Arduino already used for other GPIO purposes. You'd have to swap / remap it with POT_CS (pin A4) - and that requires software changes.
  * new code to control the PWM/VPP would have to be written
  * you'd need external power, presumably 19-20V DC. That seems to me more expensive than a digi pot IC. If it is a very cheap PSU, it might not have a transformer, and that might fry your PC USB port when you connect both the external power and USB serial from Arduino to your PC (grounds from PC and PSU on different level).

To address your questions:

> how much current would programming request? .. Do this 1A fit the bill ?

I did not measure it, but 1A seems more than enough. My guess is that VPP draws less than 250mA.

> In the code you use indexes to address wiper positions ...

Correct. In your case each index would correspond to a different duty cycle of PWM (0 to 255 in 8bit PWM)

>  Found voltages ranging from 5V ~ 19V ., .. would there be more?

Actually there is less than that. You'd need:
* 4.75V to 5V VCC (any value would do)
* 10V to 16.5V in 0.5V steps. Ideally the GAL IC's specify the VPP voltage in 0.25V steps, but so far it seems that
  0.5V steps are sufficient.
  
If you are looking for an existing solution that drives the VPP via PWM, check GALMate hardware here:
https://www.ythiee.com/2021/06/06/galmate-hardware/

> ..and the slew rate of the ramp-up?

Afterburner's VPP ramps up from 5V to 12V in ~ 1mS. This is mainly to prevent the voltage booster to overhoot the target voltage. 

Hope it helps.

fredcwbr, 2024-02-13:
Hi, and thanks for the reply.

> I'm sorry to read the MCP hardware is not available where you live.
If it would be sourced, there is a minimum order of 100 parts... and shipping would be around the same amount. I tried on Mouser, Fornell, Alibaba, AliExpress  and others.  The overall cost would be almost equivalant to a Xgecu;; 

> Unless you want to go for a long journey or redesigning Afterburner's software and hardware ......
> want to program few GAL16V8 or GAL22V10 (or their Atmel equivalent)
I'm considering. Although you pointer out two major goals, 16V8s and 22V10s

>  use the legacy hardware design 
Was already looking into it

>  want it for an educational purpose, in a electronic club or hacker space?
Yes, on the long run, after getting myself working with it. Not new to electronics., but GALs, CPLDs and FPGA banged on my door.

> really need programmable VPP, then buy MCP4151-104e
I found other pots available that i could bodge a connection, ike the MCP4151, but reading your code, noticed that you read the pots value. Is this for the autodetection? .. 

>  AD8400ARZ10 --
Same issue as others, can only order a reel

> If you really would like to use the solution you suggested, here are the issues I can see:
No I don't. Actually I stole  the idea from another project, - GALMate - it uses that solution (in a way)
with a DC-DC converter , and the PWM (not  that creative here)  8-) .

> mapping/remapping, recoding
Well no problem there. Thats my bread and butter., got into this mess for the need do  make an overhaul of a pice of 1970s equipment, with NO docs (great black box), NO schematic, No standards, .. 
so its already been over the bench about 3 months now. Got a solution, and will go FPGA so deliveries can be done, but it was annoying havig to get a 240LU FPGA just to do what 8 TTL logics are doing. It hurts.

> To address your questions:
Thank I'll listen carefully

>> 4.75V to 5V VCC (any value would do)
OK gotcha...
>> 10V to 16.5V in 0.5V steps. Ideally the GAL IC's specify the VPP voltage in 0.25V steps, but so far it seems that 0.5V steps are sufficient.
Done!!!

>  VPP via PWM, check GALMate hardware here: https://www.ythiee.com/2021/06/06/galmate-hardware/
Yeah ... stole the hole idea from him... 

> Afterburner's VPP ramps up from 5V to 12V in ~ 1mS.
It's in my ballpark.

> Hope it helps.

More than you can imagine.

Best Regards from Sen. Canedo - Goias - Brasil

closed 2024-02-13 by fredcwbr

ole00, 2024-02-14:
Hi Frederico, I have a spare MCP4131 IC. I'm happy to send you one if you want to. If so, contact me via email, issue 'git log' command in the Afterburner's directory to find it.

fredcwbr, 2024-02-15:
Howdy,

Dropped the note there. 
Thanks in advance.

and
Best Regards.

fredcwbr, 2024-02-15:
Howdy,

I'ved followed your advice on the legacy version,. and it worked for my immediate need. 
As  i'm also closely following the VPP issue with HuberH, decided to dig into legacy and adapt a power source according to that. 
Just cloned the legacy branch and will work towards that .,.  will request a pull in due time.

fredcwbr, 2024-02-16:
Howdy,
After careful reading on the NewVPP proposed by HuberH , and looking into the original  V3, it occured to me that your using 4131 as a reostat to GND. 
Did a try with a  PCF and a R2R 8 bit resistor DAC , and it worked the same. 
Can still read the PCF , and using OC, it does not interfere with the boost. 

So i can make a piggyback on the  DIP8 3141 socket and put whats needed there to emulate. Just to stay true to V3 for now.

ole00, 2024-02-16:
Yes, MCP4131 works as a rheostat. The resistor 'DAC' could possibly be used along with another 595 shift register (untested). The idea is that the resistors could be connected to the outputs of the 595 shift register. If the shift register value would be 0, then all resistors would conduct to GND (low pot value). If value 255 would be set to the shift register, then the resistors would not conduct (maximum resistance - actually this situation would have to be handled by another resistor in parallel, so that the total Feedback resistance would be ~25k Ohm). Any other value would pass some current through resistors (where the bits on 595 outputs are set to 0).  A clever selection of 8 (or 9) resistors would have to be calculated so that their parallel combinations would give a value between 10k to 25k.

ole00, 2024-02-16:
The downside of the above solution is that the 595 has a bigger footprint and also it would require more resistors to fit on the PCB. Therefore the Afterburner's PCB area would have to be bigger (if through hole components are required). The R9 pot could possibly be removed as the new rheostat could possibly be more precise (no need for the POT compensation ins such case). 
Actually, MCP4131 has only 128 taps, therefore only 7 outputs from the 595 would be needed.

fredcwbr, 2024-02-16:
Keeping the footprint inside arduinos shield factory format is a need... did i understand this correctly?   Any other concerns?