ole00/afterburner #54: Redesigned algorithm for VPP measurement

hubertushirsch, 2024-02-14:
I propose a simplified form of voltage measurement in the Afterburner.
This affects the **varVppMeasureVpp()** function in **aftb_vpp.h** and some definitions.
The mathematical basics:

ADC = return value of analogRead()
Vin = voltage at the analog input
res = resolution - 2^10 = 1024 (10 bit resolution)
Vpp = programming voltage VPP
Vref = reference voltage of the A/D converter at pin AREF

Based on the basic formula
```
(1) ADC = Vin * Res / Vref
```
After converting formula (1) we get
```
(2) Vin = ADC * Vref / res
```
**Vin** results from the voltage divider R5/R6
```
Vpp ---+
       |
       -
      | | R5
      | |
       -
       |
       o---- Vin to analog input
       |
       -
      | | R6
      | |
       -
       |
      ___ GND
      
(3) Vpp / (R5 + R6) = Vin / R6
```
Switched to **Vpp**
```
(4) Vpp = Vin * (R5 + R6) / R6
```
**Vin** substituted by (2):
```
(5) Vpp = (ADC * Vref * (R5 + R6)) / (R6 * res)
```
If you calculate with floating point, you can measure the Vref of your specific afterburner with a multimeter and specify it precisely, e.g.
```
#define VREF (3.28)      // Vref in Volts as float
```
This also applies to the resistors R5 and R6, which can be measured beforehand.
```
#define ADC_R5 (99.96)   // R5 in kOhm as float
#define ADC_R6 (20.01)   // R6 in kOhm as float
```
I put the definitions of the parameters in a separate header file **aftb_adcparms.h**, which is included in **aftb_vpp.h**.
This makes it easy to adapt to your own hardware without having to spend a long time searching between the lines of code.
The **varVppMeasureVpp()** function has been completely rewritten.

Measuring using this method provides a significantly better agreement between the displayed VPP values compared to a multimeter.
The difference between the Arduino measurement and the multimeter was a maximum of 0.02V (Arduino NANO, Vref = internal 1.1V, R5 = 20k, R6 = 1.3k).

hubertushirsch, 2024-02-15:
> I'm following this with great interest., ..

I would be very grateful if you could test the suggested code change.
See pull request https://github.com/ole00/afterburner/pull/55

fredcwbr, 2024-02-15:
I'll look into it closely, at the time i can get my hands in a 4131..

Emile666, 2024-09-23:
So I built an afterburner and started by checking Vpp voltages. I was amazed by how inaccurate the readings were: A Vpp of 16.50 Volt (multimeter) resulted in an reported voltage of 16.08 Volts. I observed the following:
- The AREF pin is not decoupled
- There is a series resistor (R7) of 3K3 from the +3V3 to AREF. The AREF pin has a current draw of around 94 uA (measurement and confirmed by graph in datasheet). This results in a VREF of 3.04 Volts instead of 3.30 Volts.
- Pot R9 doesn't decrease the inaccuracy between the voltmeter readings and the values on screen.
- The Vpp reading is read in 14 times and then divided by 8 (no comment why it is done this way). A VPP of 16.50 then results in a code of (16.5 / 6) * (14/8) * 1023 / 3.04 = 1619, which is pretty much the voltage reported on screen. So this is by design and it is way off from the actual value!

- So there are a couple of hardware improvements to make: decouple the AREF pin with 100 nF and either remove R7 completely or make the value a lot smaller.  
- The Full-Scale for the ADC is 1980 E-2 V (this is 6 * 330 E-2 V, the 6 is the R6/R5 voltage divider). Therefore, the ADC value is multiplied with 1980 and divided by n*1023, n being the number of times the ADC value is read. Create a #define for this:
#define R5 (100.0) /* value in kΩ */
#define R6 (20.0) /* value in kΩ */
#define VDIV (1.0+(R5/R6))
#define ADC_AVG (14) /* number of times ADC value is read in */
#define MAX_ADC_VAL (1023) /* 2^10 - 1 */ 
#define VREF (330) /* E-2 V */
#define ADC_MUL_VAL ((VDIV * VREF)/(ADC_AVG*MAX_ADC_VAL) )
If you then multiply (not divide, too computation intensive!) the 14-times summed ADC value with ADC_MUL_VAL, you end up with the proper value in E-2 V. Note that by using these defines, calculation is transparent and the values are easy to change.
- I would support the software changes suggested by hubertushirsch, since that is basically what I am describing here.