ole00/afterburner #59: Incomplete output from PC program

hubertushirsch, 2024-03-18:
If I enable **VPP_VERBOSE** in **aftb_vpp.h** line 44
```
#define VPP_VERBOSE 1
```
then the output of the command in the PC program is cut off.
```
D:\<path> .\afterburner_w64.exe m -d COM15

VPP calib. offset: 0
VPP: 4.2 - 5.0V : varSetVppIndex 0
4.74
VPP: 9.0V : varSetVpp 0:44
varSetVppIndex 20
varSetVppIndex 44
9.01
VPP: 10.0V : varSetVpp 2:106
varSetVppIndex 51
varSetVppIndex 75
varSetVppIndex 87
varSetVppIndex 93
varSetVppIndex 96
varSetVppIndex 99
varSetVppIndex 102
varSetVppIndex 105
varSetVppIndex 106
10.01
VPP: 12.0V : varSetVpp 6:180
varSetVppIndex 88
varSetVppIndex 112
varSetVppIndex 124
varSetVppIndex 130
varSetVppIndex 133
varSetVppIndex 136
varSetVppIndex 139
varSetVppIndex 142
varSetVppIndex 145
varSetVppIndex 148
varSetVppIndex 151
varSetVppIndex 154
varSetVppIndex 157
varSetVppIndex 160
varSetVppIndex 163
varSetVppIndex 166
varSetVppIndex 169
varSetVppIndex 172
varSetVppIndex 175
varSetVppIndex 178
varSetVppIndex 180
12.01
VPP: 14.0V : varSetVpp 10:222
varSetVppIndex 109
varSetVppIndex 133
varSetVppIndex 145
varSetVppIndex 151
varSetVppIndex 154
varSetVppIndex 155
varSetVppIndex 156
varSetVppIndex 157
varSetVppIndex 158
varSetVp               <== cut off here
D:\<path>
```
Running the **m** command on the serial terminal, the output is complete.
```
MCP4151 POT found
AFTerburner v.0.5.8
 varVpp
type 'h' for help
I: VPP calib. OK
>
m
VPP calib. offset: 0
VPP: 4.2 - 5.0V : varSetVppIndex 0
4.75
VPP: 9.0V : varSetVpp 0:44
varSetVppIndex 20
varSetVppIndex 44
9.01
VPP: 10.0V : varSetVpp 2:106
varSetVppIndex 51
varSetVppIndex 75
varSetVppIndex 87
varSetVppIndex 93
varSetVppIndex 96
varSetVppIndex 99
varSetVppIndex 102
varSetVppIndex 105
varSetVppIndex 106
10.01
VPP: 12.0V : varSetVpp 6:180
varSetVppIndex 88
varSetVppIndex 112
varSetVppIndex 124
varSetVppIndex 130
varSetVppIndex 133
varSetVppIndex 136
varSetVppIndex 139
varSetVppIndex 142
varSetVppIndex 145
varSetVppIndex 148
varSetVppIndex 151
varSetVppIndex 154
varSetVppIndex 157
varSetVppIndex 160
varSetVppIndex 163
varSetVppIndex 166
varSetVppIndex 169
varSetVppIndex 172
varSetVppIndex 175
varSetVppIndex 178
varSetVppIndex 180
12.01
VPP: 14.0V : varSetVpp 10:222
varSetVppIndex 109
varSetVppIndex 133
varSetVppIndex 145
varSetVppIndex 151
varSetVppIndex 154
varSetVppIndex 155
varSetVppIndex 156
varSetVppIndex 157
varSetVppIndex 158
varSetVppIndex 159
varSetVppIndex 160
varSetVppIndex 161
varSetVppIndex 162
varSetVppIndex 163
varSetVppIndex 164
varSetVppIndex 165
varSetVppIndex 166
varSetVppIndex 167
varSetVppIndex 168
varSetVppIndex 169
varSetVppIndex 170
varSetVppIndex 171
varSetVppIndex 172
varSetVppIndex 173
varSetVppIndex 174
varSetVppIndex 175
varSetVppIndex 176
varSetVppIndex 177
varSetVppIndex 178
varSetVppIndex 179
varSetVppIndex 180
varSetVppIndex 181
varSetVppIndex 182
varSetVppIndex 183
varSetVppIndex 184
varSetVppIndex 185
varSetVppIndex 186
varSetVppIndex 187
varSetVppIndex 188
varSetVppIndex 189
varSetVppIndex 190
varSetVppIndex 191
varSetVppIndex 192
varSetVppIndex 193
varSetVppIndex 194
varSetVppIndex 195
varSetVppIndex 196
varSetVppIndex 197
varSetVppIndex 198
varSetVppIndex 199
varSetVppIndex 200
varSetVppIndex 201
varSetVppIndex 202
varSetVppIndex 203
varSetVppIndex 204
varSetVppIndex 205
varSetVppIndex 206
varSetVppIndex 207
varSetVppIndex 208
varSetVppIndex 209
varSetVppIndex 210
varSetVppIndex 211
varSetVppIndex 212
varSetVppIndex 213
varSetVppIndex 214
varSetVppIndex 215
varSetVppIndex 216
varSetVppIndex 217
varSetVppIndex 218
varSetVppIndex 219
varSetVppIndex 220
varSetVppIndex 221
varSetVppIndex 222
14.00
VPP: 16.0V : varSetVpp 14:250
varSetVppIndex 123
varSetVppIndex 147
varSetVppIndex 159
varSetVppIndex 165
varSetVppIndex 168
varSetVppIndex 169
varSetVppIndex 170
varSetVppIndex 171
varSetVppIndex 172
varSetVppIndex 173
varSetVppIndex 174
varSetVppIndex 175
varSetVppIndex 176
varSetVppIndex 177
varSetVppIndex 178
varSetVppIndex 179
varSetVppIndex 180
varSetVppIndex 181
varSetVppIndex 182
varSetVppIndex 183
varSetVppIndex 184
varSetVppIndex 185
varSetVppIndex 186
varSetVppIndex 187
varSetVppIndex 188
varSetVppIndex 189
varSetVppIndex 190
varSetVppIndex 191
varSetVppIndex 192
varSetVppIndex 193
varSetVppIndex 194
varSetVppIndex 195
varSetVppIndex 196
varSetVppIndex 197
varSetVppIndex 198
varSetVppIndex 199
varSetVppIndex 200
varSetVppIndex 201
varSetVppIndex 202
varSetVppIndex 203
varSetVppIndex 204
varSetVppIndex 205
varSetVppIndex 206
varSetVppIndex 207
varSetVppIndex 208
varSetVppIndex 209
varSetVppIndex 210
varSetVppIndex 211
varSetVppIndex 212
varSetVppIndex 213
varSetVppIndex 214
varSetVppIndex 215
varSetVppIndex 216
varSetVppIndex 217
varSetVppIndex 218
varSetVppIndex 219
varSetVppIndex 220
varSetVppIndex 221
varSetVppIndex 222
varSetVppIndex 223
varSetVppIndex 224
varSetVppIndex 225
varSetVppIndex 226
varSetVppIndex 227
varSetVppIndex 228
varSetVppIndex 229
varSetVppIndex 230
varSetVppIndex 231
varSetVppIndex 232
varSetVppIndex 233
varSetVppIndex 234
varSetVppIndex 235
varSetVppIndex 236
varSetVppIndex 237
varSetVppIndex 238
varSetVppIndex 239
varSetVppIndex 240
varSetVppIndex 241
varSetVppIndex 242
varSetVppIndex 243
varSetVppIndex 244
varSetVppIndex 245
varSetVppIndex 246
varSetVppIndex 247
varSetVppIndex 248
varSetVppIndex 249
varSetVppIndex 250
16.04
varSetVppIndex 0
>
```
Regards
Hubert

hubertushirsch, 2024-03-18:
The same effect with other commands with long outputs.
The output of **.\afterburner_w64.exe b** is truncated,
**b** on the serial terminal produces the full output.

ole00, 2024-03-19:
Thanks for the report. I'll check that. Hopefully it is something minor. I suspect the the delay in the PC program is not long enough.

ole00, 2024-03-19:
It's now fixed (see commit 79e33ab2a5963c421ecf021ac208408b4c74ef43  for more information). I've updated the PC app releases just in case you do not compile the Windows version of Afterburner PC app yourself.

hubertushirsch, 2024-03-19:
Thanks, it seems to be working.

Hubert

closed 2024-03-19 by hubertushirsch