ole00/afterburner #61: ZIF Socket is not powerless when removing a GAL

hubertushirsch, 2024-03-21:
I have observed that when removing a GAL (currently only have ATF16V8B) the LED sometimes flashes.
This depends on how I move the chip when removing it and in which order the pins lose contact on the ZIF socket.
Since the Powerswitch SW1 is switched off, the current through the LED must come from another connection on the ZIF socket that has a HIGH level and flow through the inactive GAL chip.
In fact, ZIF16 (Arduino D9, DOUT for this GAL type) is set to HIGH due to the pin mode on D9
```((pinMode(PIN_ZIF16, INPUT_PULLUP); //DOUT**)```
after the first command.
A test shows: When connecting a wire from ZIF16 to ZIF15, the LED lights up.

I don't know whether this could destroy the GAL when plugging or removing it in the worst case scenario.
Is it possible to change the sketch so that there is no HIGH level at the DOUT of the last selected GAL type if the GAL is not addressed by a command?
The best would be a pulldown resistor, but this cannot be programmed on the Arduino input.
The alternative would be INPUT without pullup, but that would be a floating input.

ole00, 2024-03-22:
Well spotted. Yes, it is a bug - the DOUT pin (output from GAL, input to arduino)  should be set to input , not input pull-up when inactive.  Ideally all pins leading to the ZIF socket should be completely isolated (because even a pin set to input can damage Arduino's gpio circuitry), but that would require more ICs on the board leading to higher board complexity and higher cost.
It is now fixed in commit 1589119e7c4f144fd4dca17e199bb6b854f70574
Thanks for the report.

hubertushirsch, 2024-03-22:
Thanks for the fix, seems to be working.

Hubert

closed 2024-03-22 by hubertushirsch