ole00/afterburner #66: Afterburner seems to destroy GAL

Paul-AGI, 2024-08-31:
I built the latest version of the afterburner and I'm trying to use it to program some Lattice GAL16V8B chips.  I'm using one of the newer MT3608 modules where I had to replace the 2.2K resistor with a 15K resistor.  After that, callibration went fine.

Everything seems fine, putting the chip in with the i command gets me

$afterburner i -t GAL16V8
PES info: 5V Lattice GAL16V8  VPP=12.0 Timing: prog=40 erase=25

I can erase it without an error, but after writing a JED file produced by CUPL with a g16v8 device definition, I get this:
$ afterburner wv -t GAL16V8 -f STATEMAC.JED
Uploading fuse map...
2194/2194 |########################################|
ER unknown or wrong GAL type (check Power ON)

then when trying an i command again, I get this:
$ afterburner i -t GAL16V8
PES info: 5V Unknown GAL,  try VPP=10..14 in 1V steps

This does the same behavior with a Lattice GAL16V8D as well

Any clues on where to look on this?  I didn't use a PCB but it follows v3.1 of the schematic.

ole00, 2024-09-02:
Hi Paul,

sorry to hear your Afterburner does not work as intended. You mentioned you did not use Afterburner PCB, did you use breadboard? Or did you use veroboard? Could you send a picture of your Afterburner (ideally both sides - if you use some board to solder parts)? Also, you mentioned the calibration was OK. Could you run the afterburner's 'm' command again to measure voltages (while no GAL chip is inserted) to verify the MT3608 module is still working OK?
With the new Afterburner design (with the VPP voltage booster) the errors you described in general are:
- the VPP does not work correctly (hence the new VPP measurement with the 'm' command),
- or the is a soldering mistake (a bridge, or wrong resistor value or bad solder joint)
- ZIF socket contacts are worn out, therefore a pin can have a bad/unstable connection (reseating the IC or pushing down on it while reading the GAL might help)

Paul-AGI, 2024-09-03:
I'm sure I have a mistake somewhere.

The m command returns:
afterburner m

VPP calib. offset: 5
VPP: 4.2 - 5.0V : 4.69
VPP: 9.0V : 8.99
VPP: 12.0V : 11.97
VPP: 14.0V : 13.99
VPP: 16.0V : 15.99

Those values are within +/- .02V of actual.

Doing a read with -nc after writing  shows no change to the fuse map but a zeroed PES.  Attempting to write the PES results in no change.  The programming voltage during writing was 11.95.  I used a new zif socket (although it's probably a counterfeit).

The first write on a new 16V8 works, but overwrites the pes so the verify fails.  The erase and read work before writing.

This was breadboarded using a Nano and everything has been triple checked, but clearly I'm missing something.  I'll put a LA on the chip and see if the programming looks reasonable; maybe I can catch what's going on.  I did find the GALblast description of the programming algorithm so that might help me see what's going on.

Paul-AGI, 2024-09-04:
I pulled the gal and hooked up a LA.  This is the capture I get with this command

afterburner p -t GAL16V8 -pes 00:05:1A:A1:CD:A4:03:06  -nc

Although it takes up the complete screen, that is the complete capture, there is no activity after that.

[image]

In particular, the SDIN and SCLK do not look right to me.  This should be clocking in 64 bits, right?

ole00, 2024-09-09:
Perhaps I don't read your capture screen shot correctly, but it seems to me the sample rate is one sample every 4 ms, which is 250Hz. That is not fast enough to read the Afterburner data properly. Try to increase sample rate to at least 1MHz to check whether you can get better clock and data values in your logic analyser.