ole00/afterburner #67: Couple of bugs found during code review

Emile666, 2024-09-30:
While reviewing the code, I found a number of (potential) bugs:
1) L118 & L119: the cfgbits field is not filled in for GAL26CV12 and GAL26V12. I don't know why I don't get a warning from the compiler, strange. But there really is one data byte missing.
2) L638: if (verbose & bigRam), this should be a double & (verbose && bigRam). A single & is a bitwise operator
3) L1151: the function sendLine expects an integer for the 2nd parameter (bufSize), but it is given the value of GALBUFSIZE, which is 256 * 1024 or an uint32_t. It now overflows and probably passes a value of 0.

Please advice how to solve these.

Emile666, 2024-10-01:
If the config bits are the S0, S1 etcetera bits in the macro-cells, then the values for the cfg-bits for GAL26CV12 and GAL26V12 are 24 (S0,S1: 2x12 macro-cells) and 48 (S0..S3: 4x12 macro-cells) respectively. Please confirm if this is correct.

I did set GALBUFSIZE to (31 * 1024) just to get rid of this bug. But don't know how much it should be. If it needs to be 256 * 1024, then the function itself has to change, so that bufSize can be an uint32_t.

Emile666, 2024-10-02:
Thanks very much for answering!

When reporting issue 3, I clearly made a mistake by reporting the wrong line-numbers, my bad! The define GALBUFSIZE is defined at line 62 and is used at **line 1221**: "readSize = sendLine(buf, GALBUFSIZE, 12000);"
Since the 2nd parameter of sendLine is an integer, this will create an overflow.

I will update my post above to correct the linenumbers.

Emile666, 2024-10-02:
Another issue I came across, is the VPP measurement. I added a detailed analysis to issue #54 already reported by HubertusHirsch. I largely agree with him and found the same (inaccurate) measurements. I also added some code-improvements on how to resolve this. Readings are now accurate within 0.05 Volts and can be made even smaller without adding offsets.

Will you please have a look at my comments and proposal in issue #54? I am happy to help with a pull-request or sending in source files if you wish.

ole00, 2024-10-03:
> Since the 2nd parameter of sendLine is an integer, this will create an overflow.

I see, you mean in a case where the 'int' is defined as 16 bit value? I don't think any compiler for PC-as-a-target does that. Afterburner.c is compiled for a PC, not for Arduino or other microcontroller, so that should not be an issue. Or am I missing something? Do you compile Afterburner.c for a microcontroller?

VPP measurements: the GALS are not that VPP sensitive. I had no issue calibrating the VPP to +/- 50 mV with the existing algorithm. If you prefer the new VPP measurement algorithm feel free to use it. I prefer the VPP measurement algorithm the way it is now.

Emile666, 2024-10-04:
When compiling on a Windows-11 PC with GCC and mingw, I got the following warning: "overflow in conversion from 'int' to 'int16_t' {aka 'short int'} changes value from '262144' to '0'. It is standard programming practice not to rely on the size of an int, since this is machine-dependent. As an embedded engineer, I am used to an int being a 16-bit number, but again this is machine-dependent. So yes, it is a bug and the best thing to do is to use the type-definitions from stdint.h, like uint16_t or uint32_t. Only then it is clear what you mean and it will clear up the code as well. Because now it is still unclear for me if the large constant (256 * 1024) is really necessary or that I should change the sendLine function. If you would have used the types from stdint.h, this would never have been an issue.

The other thing about the Vpp measurement is even worse. You are not converting the Vpp voltage in a correct way and the R7 resistor has too much voltage-drop, so AREF gets an incorrect value. You will notice that you need larger offsets for other values, because the current conversion from analog to digital, basically the multiplication factor, is incorrect. You added lots of code with offsets just to mask the root-cause, which is that the conversion itself is not correct. If you do it properly, you don't need all the patching with offsets. What I (and Hubertushirsch as well) did suggest is a standard way of converting analog values to digital counterparts, with a proper full-scale value that corresponds with AREF. And with that it is easy to mask the AREF / R7 hardware fault, just by changing the AREF constant. I will probably change the value of R7 to something like 220 Ω and add some decoupling for AREF, so that this is solved as well.

But again, it is your software. I just wanted to point out some things to improve the overall quality of your code and to report a few bugs. I did correct them in my local repository and wanted to share that with you. But if there's no interest for that, it is fine by me. I still like your gal programmer and thank you for creating such a nice little system!

ole00, 2024-10-04:
Thanks for the info. Could you please share your GCC version (gcc -v) and your compilation/linking parameters? Thanks.

Emile666, 2024-10-05:
gcc -v: Target: i686-w64-mingw32, gcc version 14.2.0 (i686-posix-dwarf-rev0, Built by MinGW-Builds project
Parameters: -g3 -o0 -o afterburner.exe afterburner.c -D_USE_WIN_API_ _DNO_CLOSE

ole00, 2024-10-06:
Thanks. I've downloaded and installed Mingw GCC 14.2.0 from here:
https://winlibs.com/
I used the following download link: [GCC 14.2.0 (with POSIX threads) + LLVM/Clang/LLD/LLDB 18.1.8 + MinGW-w64 12.0.0 UCRT - release 1 ](https://github.com/brechtsanders/winlibs_mingw/releases/download/14.2.0posix-18.1.8-12.0.0-ucrt-r1/winlibs-i686-posix-dwarf-gcc-14.2.0-llvm-18.1.8-mingw-w64ucrt-12.0.0-r1.zip)

When I compiled with this mingw version I had no compilation issues, not even a warning. I checked the size of int:
`printf("sizeof int: %d\n", sizeof(int));`
And it prints the size of int is 4.
To me it looks like you have a non standard or incorrectly built compiler.

OTOH I agree with you that using implicit bit sizes of integers would resolve the compilation issue as well. 

BTW, if you want to use the NO_CLOSE preprocessor symbol, use -DNO_CLOSE, not _DNO_CLOSE compilation parameter.

Emile666, 2024-10-07:
That's interesting! I just downloaded and installed mingw, didn't look any further. The _D was a typo here, I used -D, but thanks for noting this.
Yes, the use of stdint.h (and stdbool.h, but that's another issue) throughout the code would not change any functionality, but it would clear it up. Again, I can help you with that if you like, I could make a pull request for you, OTOH it is not really a lot of work. Let me know what you want.

Other question: why the use of static in front of all functions? That's not really needed, is it? Or do you have a specific reason for this? You would use static in front of a function to restrict access to functions in the file itself. But you only have one file, right?

ole00, 2024-10-07:
Using static functions is a habbit of mine. In C, the static function functions a bit like a private method in C++, therefore it is not exposed outside of the compiled object file as an external function (can't be linked/accessed from another object file). Compilers are also able to do automatic inlining of static functions. In Afterburner's PC code it is not important (it is a single object file), but in bigger C projects I tend to use non-static functions only if the functions are part of an API.