ole00/afterburner #71: Write PES and Erase All issues with 22V10

ceteras, 2025-05-22:
I've found a few issues while working with a Lattice GAL22V10D which had its PES erased (10 FFs).
Replicating the issues will have to start with a completely erased chip.

Steps I followed are:
1. attempt to write a new PES:

`./afterburner_osx -d /dev/tty.usbmodem101 p -v -t GAL22V10 -nc -pes "00:05:49:A1:CD:A4:03:00"`

I took the PES from another chip.

This would send data in the wrong place in the chip, the result looks like this:

```
L5720 11111111111111111111111111111111111111111111*
L5764 11111111111111111111111111111111111111111111*
L5808 10101100000000000000*
N UES FF FF FF FF FF FF FF FF*
L5828 1111111111111111111111111111111111111111111111111111111111111111*
N PES FF FF FF FF FF FF FF FF FF FF*
CDB57
```

Line starting at 5808 was affected.
I fixed this by treating GAL22V10 similarly to GAL6001/2:
In `writePes()` function I've added a case to `switch(gal)`
```
    case GAL22V10:
      setRow(0);
      for (rbit = 0; rbit < 64; rbit++) {
        b = pes[rbit >> 3];
        p = b & (1 << (rbit & 0b111));
        sendBit(p);
      }
      sendBits(52, 0);
      sendAddress(6, galinfo.pesrow);
      setSDIN(0);
      break;
```
Using info from http://www.armory.com/%7Erstevew/Public/Pgmrs/GAL/algo.htm I tried adding the 52 L bits.

This caused the data to be sent to the proper row in the chip, but because the PES sent was 8 bytes not 10 as 22V10 requires, the data was shifted two bytes to the right:

```
L5764 11111111111111111111111111111111111111111111*
L5808 11111111111111111111*
N UES FF FF FF FF FF FF FF FF*
L5828 1111111111111111111111111111111111111111111111111111111111111111*
N PES 00 00 00 05 49 A1 CD A4 03 00*
CDD2F
```

2. attempt to erase the wrong PES
The erase with -all argument does not execute if a chip cannot be identified (-nc is required if you have a bad PES).

I've changed the check in afterburner.c in main:
```
if (opErase /*&& 0 == result*/) {
         result = operationEraseGal();
     }
```

Another issue seems to be the Erase All row address, in galInfoList[] I've set the `eraseallrow` to 57 for GAL22V10 (value taken from the same web page above.
I was able to erase the bad PES value.

3. Hack to shift the PES to its correct location
As we can only send 8 bytes for PES, I've changed the number of L bits sent from 52 to 68 :
```
    case GAL22V10:
      setRow(0);
      for (rbit = 0; rbit < 64; rbit++) {
        b = pes[rbit >> 3];
        p = b & (1 << (rbit & 0b111));
        sendBit(p);
      }
      sendBits(68, 0);
      sendAddress(6, galinfo.pesrow);
      setSDIN(0);
      break;
```

After these hacks I could have a proper PES written in my chip:
```
L5720 11111111111111111111111111111111111111111111*
L5764 11111111111111111111111111111111111111111111*
L5808 11111111111111111111*
N UES FF FF FF FF FF FF FF FF*
L5828 1111111111111111111111111111111111111111111111111111111111111111*
N PES 00 05 49 A1 CD A4 03 00 00 00*
CDD2F
```

... and identify it:

```
./afterburner_osx -d /dev/tty.usbmodem101 i -t GAL22V10   
PES info: 5V Lattice GAL22V10  VPP=13.0 Timing: prog=40 erase=25
```

ole00, 2025-05-22:
Thanks for the research and for sharing the information along with your code snippets. I have some GAL22V10D and GAL22V10B available, so will check your solution on both chips and update the code soon.

ole00, 2025-05-22:
BTW. How did you find out you need to change the number of bits from 52 to 68 in "sendBits(68, 0);" ? Experimentally? Or have you found any note regarding this ? The link on the armory.com documentation mentions 80 bits for PES data on GAL22V10. Thanks.

ceteras, 2025-05-23:
The armory page mentions 52 bits, and I needed two extra bytes for the PES (I've sent an 8 byte argument as it's the only accepted length).
The 16 extra bits compensate for the PES argument not being 10 bytes long, and are actually part of the PES, in the end.
I was lucky that the last 3 bytes in the PES are all zeroes.
If you change the pc app to accept 10 bytes PES, then for 22V10 only 52 low bits are needed.

ceteras, 2025-05-23:
Another issue I've found related to bad PES in the chip: when you want to read the chip and your PES is bad, you have to use the "-nc" flag, and this means the `pes` variable will not be initialized with the actual wrong value in the chip, but with 12 zeroes (by `case COMMAND_DISABLE_CHECK_TYPE`).
So `printJedec()` will print zeroes for the PES section, and we cannot see the actual values stored .

My workaround was to add a call to `readpes()` in `printJedec()` function:

```
    readPes();
    Serial.print(F("N PES"));
```

ceteras, 2025-05-23:
Thank you, I'll give it a try!

ceteras, 2025-05-24:
It didn't work for my chips, however I don't trust the source I've got them from.
I'm waiting for an order of 4x 26CV12 from another seller, I'll let you know how that works.

Question, why the 86 for the GAL26 ?

PES: RA=0:  b0..b95, 26*L, L, H, L, H, H, H (96+26+6 bit, line 58)
That is 128 bits in total.
So, 64bits PES, 86 bits L, 6 bits addres is 156.
For 26CV12 it should be : 128 - 64 - 6 = 58 L bits.
Anyway, with both options, my chips are completely inert, one of them returns all zeroes, the rest return all ones, when read.
Will let you know when I try the new chips.

ceteras, 2025-06-20:
I've had the change to test with good working Lattice GAL26CV12 chips.
This is the reason I've chosen afterburner: support for the GAL26 types, on an open source project.

 The good news is the programmer worked from the start with all 4 of them, and the only change needed for complete support is the change suggested by you for the writePes, that is sending 58 zeroes.

```
    case GAL26CV12:
      setRow(0);
      for (rbit = 0; rbit < 64; rbit++) {
        b = pes[rbit >> 3];
        p = b & (1 << (rbit & 0b111));
        sendBit(p);
      }
      sendBits(58, 0);
      sendAddress(6, galinfo.pesrow);
      setSDIN(0);
      break;
```

There is one more thing I've noticed again while testing:

There is no way of inspecting the PES contents of a chip when the PES is corrupted.
Eg. When I've erased the PES with " e -all ", I knew its contents were FF, but command "r -nc" shows zeroes (that is the init value for the pes array in code).
Expected would be full FF (12 bytes for this case).
Then if you work on the writePes routine and you didn't get it just right, and your PES is shifted, so still bad, then you still need to use -nc for reading, and still get "00" 12 times.
Expected would be the wrong actual PES value.

How did you inspect the PES while testing ?

My workaround was to squeeze a call to readPes() in  readGalFuseMap function, like this:

```
// generic fuse-map reading, fuse-map bits are stored in fusemap array
static void readGalFuseMap(const unsigned char* cfgArray, char useDelay, char doDiscardBits) {
  unsigned short cfgAddr = galinfo.cfgbase;
  unsigned short row, bit;
  unsigned short addr;
  readPes();
  if (flagBits & FLAG_BIT_ATF16V8C) {
      setPV(0);
  }
```

Then when I read the gal with "r -nc" I get the actual PES values.

Thanks again for this project and your support!

ole00, 2025-06-21:
Thanks for the feedback and verification. It came with good timing as I've bough GAL26CV12B (which was missing in my GAL collection) to do further checks (I did not know at that time that you'd still work with GAL26CV12). The only GAL26CV12 chips  available in my region for purchase are in PLCC28 package, so I made a PLCC28 adapter for Afterburner. I guess the adapter might get handy for other people so I shared it in my latest commit. 

[image]

> How did you inspect the PES while testing ?

There is a #define DEBUG_PES in afterburner.ino, which is normally disabled. If you enable it, recompile code and upload to Arduino then all afterburner operations will print the PES in the PC app log.

> Thanks again for this project and your support!

I'm glad it works for you. The project is based on work of several other people sharing their code and knowledge, so big thanks to them as well.